orbbec/OrbbecSDK_ROS2 #2: ros2 launch orbbec_camera astra.launch.xml  had crash！

dongpan90, 2023-06-05:
[crash]
[orbbec_camera_node-1] [INFO] [1685974733.694983242] [camera.camera]: try to disable ir stream and try again
[orbbec_camera_node-1] [INFO] [1685974733.695057003] [camera.camera]: Enable depth stream
[orbbec_camera_node-1] [INFO] [1685974733.695111148] [camera.camera]: Stream depth width: 640 height: 480 fps: 30 format: Y11
[orbbec_camera_node-1] [INFO] [1685974733.695162221] [camera.camera]: Enable color stream
[orbbec_camera_node-1] [INFO] [1685974733.695209229] [camera.camera]: Stream color width: 640 height: 480 fps: 30 format: RGB
[orbbec_camera_node-1] [ERROR] [1685974733.816873091] [camera.camera]: checkConnectTimer: device  not connected
[orbbec_camera_node-1] [ERROR] [1685974734.816828259] [camera.camera]: checkConnectTimer: device  not connected
[orbbec_camera_node-1] terminate called without an active exception
[orbbec_camera_node-1] *** Aborted at 1685974735 (unix time) try "date -d @1685974735" if you are using GNU date ***
[orbbec_camera_node-1] PC: @                0x0 (unknown)
[orbbec_camera_node-1] *** SIGABRT (@0x3e8000014d7) received by PID 5335 (TID 0xffff6db4a300) from PID 5335; stack trace: ***
[orbbec_camera_node-1] [ERROR] [1685974735.819567342] [camera.camera]: checkConnectTimer: device  not connected
[orbbec_camera_node-1]     @     0xffff864927c0 ([vdso]+0x7bf)
[orbbec_camera_node-1]     @     0xffff85b95d78 gsignal
[orbbec_camera_node-1]     @     0xffff85b82aac abort
[orbbec_camera_node-1]     @     0xffff85d9a83c __gnu_cxx::__verbose_terminate_handler()
[orbbec_camera_node-1]     @     0xffff85d9821c (unknown)
[orbbec_camera_node-1]     @     0xffff85d98280 std::terminate()
[orbbec_camera_node-1]     @     0xffff863b82c0 std::_Sp_counted_ptr_inplace<>::_M_dispose()
[orbbec_camera_node-1]     @     0xaaaac59c2ef0 std::_Sp_counted_base<>::_M_release()
[orbbec_camera_node-1]     @     0xffff863cb474 orbbec_camera::OBCameraNode::OBCameraNode()
[orbbec_camera_node-1]     @     0xffff863afa98 orbbec_camera::OBCameraNodeDriver::initializeDevice()
[orbbec_camera_node-1]     @     0xffff863b6e84 orbbec_camera::OBCameraNodeDriver::startDevice()
[orbbec_camera_node-1]     @     0xffff863b73b4 orbbec_camera::OBCameraNodeDriver::onDeviceConnected()
[orbbec_camera_node-1]     @     0xffff863b7e54 orbbec_camera::OBCameraNodeDriver::queryDevice()
[orbbec_camera_node-1]     @     0xffff85dc5fac (unknown)
[orbbec_camera_node-1]     @     0xffff85966624 start_thread
[orbbec_camera_node-1]     @     0xffff85c3349c (unknown)

[mydevice]: Astra pro
#lsusb
Bus 001 Device 009: ID 2bc5:0403 VIA Labs, Inc.          USB2.0 Hub             
Bus 001 Device 007: ID 2bc5:0501  
Bus 001 Device 005: ID 05e3:0610 Genesys Logic, Inc. 4-port hub

# ls -al /dev/astra_pro 
lrwxrwxrwx 1 root root 15 Mar 15 23:04 /dev/astra_pro -> bus/usb/001/009

jian-dong, 2023-07-13:
The current SDK is not adapted to Astra pro.

delihus, 2023-07-16:
> The current SDK is not adapted to Astra pro.

Please add this to the README.md