orbbec/OrbbecSDK_ROS2 #3: gemini [d2c_viewer]: rgb and depth image size not match(1920, 1080) vs (1280, 800)

TonyTnT, 2023-07-12:
device: gemini
[image]
ros2: galactic
launch config:
```
    <arg name="depth_registration" default="true"/>
    <arg name="serial_number" default=""/>
    <arg name="device_num" default="1"/>
    <arg name="vendor_id" default="0x2bc5"/>
    <arg name="product_id" default=""/>
    <arg name="enable_point_cloud" default="true"/>
    <arg name="enable_colored_point_cloud" default="true"/>
    <arg name="point_cloud_qos" default="default"/>
    <arg name="connection_delay" default="100"/>
    <!-- RGB -->
    <arg name="color_width" default="1920"/>
    <arg name="color_height" default="1080"/>
    <arg name="color_fps" default="30"/>
    <arg name="color_format" default="MJPG"/>
    <arg name="enable_color" default="true"/>
    <arg name="flip_color" default="false"/>
    <arg name="color_qos" default="default"/>
    <arg name="color_camera_info_qos" default="default"/>
    <arg name="enable_color_auto_exposure" default="true"/>

    <arg name="depth_width" default="1280"/>
    <arg name="depth_height" default="800"/>
    <arg name="depth_fps" default="15"/>
    <arg name="depth_format" default="Y11"/>
    <arg name="enable_depth" default="true"/>
    <arg name="flip_depth" default="false"/>
    <arg name="depth_qos" default="default"/>
    <arg name="depth_camera_info_qos" default="default"/>

    <arg name="ir_width" default="640"/>
    <arg name="ir_height" default="400"/>
    <arg name="ir_fps" default="30"/>
    <arg name="ir_format" default="Y10"/>
    <arg name="enable_ir" default="false"/>
    <arg name="flip_ir" default="false"/>
    <arg name="ir_qos" default="default"/>
    <arg name="ir_camera_info_qos" default="default"/>
    <arg name="enable_ir_auto_exposure" default="true"/>

    <arg name="publish_tf" default="true"/>
    <arg name="tf_publish_rate" default="10.0"/>
    <arg name="ir_info_url" default=""/>
    <arg name="color_info_url" default=""/>
    <arg name="log_level" default="none"/>
    <arg name="enable_publish_extrinsic" default="false"/>
    <arg name="enable_d2c_viewer" default="true"/>
    <arg name="enable_soft_filter" default="true"/>
```
this there any param wrong?

jian-dong, 2023-07-13:
`d2c_viewer` is for the depth color overlay viewer. This problem is depth 1280x800 can not align to color 1920x1080; if aligned success, it will output the same resolution 1920x1080.

TonyTnT, 2023-07-13:
> 

it's clear that the error msg is caused by the different resolution between rgb and depth cam, but it seems that they cannot be set to the same resolution. do I miss any info about the cam?