orbbec/OrbbecSDK_ROS2 #7: Gemini2 doesn't works

Ultramarine1939-syujie, 2023-09-01:
[ERROR] [orbbec_camera_node-1]: process has died [pid 3187, exit code -11, cmd '/home/deck/repos/Orbbec_SDK/install/orbbec_camera/lib/orbbec_camera/orbbec_camera_node --ros-args -r __node:=camera -r __ns:=/camera --params-file /tmp/launch_params_y6ics8no --params-file /tmp/launch_params_62hcctmg --params-file /tmp/launch_params_k0nd6jvu --params-file /tmp/launch_params_xtbdhoa3 --params-file /tmp/launch_params__3he3c89 --params-file /tmp/launch_params_2_bkb9pn --params-file /tmp/launch_params_era1f6ns --params-file /tmp/launch_params_wo0peeuu --params-file /tmp/launch_params_njs2hjmh --params-file /tmp/launch_params_k8xvi24l --params-file /tmp/launch_params_dn0cvhou --params-file /tmp/launch_params_lhcx9yjz --params-file /tmp/launch_params_noj81lrr --params-file /tmp/launch_params__q11w9yx --params-file /tmp/launch_params_8u93saak --params-file /tmp/launch_params_ku6ywwt3 --params-file /tmp/launch_params_v96ccdac --params-file /tmp/launch_params_4c9q81xx --params-file /tmp/launch_params_9tn1u69b --params-file /tmp/launch_params_hxr7mdty --params-file /tmp/launch_params_26j4nh3d --params-file /tmp/launch_params_vi2qtoee --params-file /tmp/launch_params_frxha27l --params-file /tmp/launch_params_8nqcr5sy --params-file /tmp/launch_params_eah2ftx2 --params-file /tmp/launch_params_36o_4r57 --params-file /tmp/launch_params_2scp0hor --params-file /tmp/launch_params_nlztp2qk --params-file /tmp/launch_params_gljvvdyg --params-file /tmp/launch_params_1c7m7qp8 --params-file /tmp/launch_params_sx5ar1os --params-file /tmp/launch_params_chcenfn2 --params-file /tmp/launch_params_b9m179c1 --params-file /tmp/launch_params_cw3rghjr --params-file /tmp/launch_params_j77gtyrb --params-file /tmp/launch_params_85wd1n9f --params-file /tmp/launch_params_evrcssw_ --params-file /tmp/launch_params__ll6_jcn --params-file /tmp/launch_params_o60f9715 --params-file /tmp/launch_params_hyj53_ff --params-file /tmp/launch_params_rj3wa3hm --params-file /tmp/launch_params_00exowoi --params-file /tmp/launch_params_b2h4fofq --params-file /tmp/launch_params_1vf8tf_d --params-file /tmp/launch_params_ixeqarnz --params-file /tmp/launch_params_b3a3z4yf --params-file /tmp/launch_params_ciowi95x --params-file /tmp/launch_params_sth39v10 --params-file /tmp/launch_params_d2e0vq9b --params-file /tmp/launch_params_mzddpei8 --params-file /tmp/launch_params__ni3mwsy --params-file /tmp/launch_params_6p48uhzr --params-file /tmp/launch_params_re6l4dvl --params-file /tmp/launch_params_g923hspx --params-file /tmp/launch_params_t18eh3nl --params-file /tmp/launch_params_3nm5kq7s --params-file /tmp/launch_params_jahgjnmu --params-file /tmp/launch_params_0vp5jz49 --params-file /tmp/launch_params_1_laijgv --params-file /tmp/launch_params__9qt_ww5 --params-file /tmp/launch_params_6wgx2gsj --params-file /tmp/launch_params_5kfxptvn --params-file /tmp/launch_params_suzw6ytb -r /camera/depth/color/points:=/camera/depth_registered/points'].

jian-dong, 2023-09-06:
@Ultramarine1939-syujie  Can you post more log before the process die?

hzwsl121, 2024-02-20:
I have a similar issue, it works fine on Ubuntu 22 on my laptop, but there are problems on Ubuntu 22 on NX.
nx@ubuntu:~/ros2_ws$ ros2 launch ros2_astra_camera astra_pro_launch.py
[INFO] [launch]: All log files can be found below /home/nx/.ros/log/2024-02-04-16-50-50-904826-ubuntu-20949
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [20961]
[INFO] [astra_camera_node-2]: process started with pid [20963]
[INFO] [uvc_camera_node-3]: process started with pid [20965]
[INFO] [robot_state_publisher-4]: process started with pid [20967]
[robot_state_publisher-4] [WARN] [1707036651.410947233] [robot_state_publisher]: No robot_description parameter, but command-line argument available. Assuming argument is name of URDF file. This backwards compatibility fallback will be removed in the future.
[robot_state_publisher-4] [INFO] [1707036651.420447271] [robot_state_publisher]: got segment camera_depth_frame
[robot_state_publisher-4] [INFO] [1707036651.420555884] [robot_state_publisher]: got segment camera_depth_optical_frame
[robot_state_publisher-4] [INFO] [1707036651.420578734] [robot_state_publisher]: got segment camera_link
[robot_state_publisher-4] [INFO] [1707036651.420595022] [robot_state_publisher]: got segment camera_rgb_frame
[robot_state_publisher-4] [INFO] [1707036651.420608079] [robot_state_publisher]: got segment camera_rgb_optical_frame
[astra_camera_node-2] [INFO] [1707036651.427359209] [astra_wrapper]: Device "2bc5/0614@2/12" found.
[astra_camera_node-2] [INFO] [1707036651.428682700] [astra_camera_node]: ir_frame_id = 'camera_ir_optical_frame'
[astra_camera_node-2] [INFO] [1707036651.428881270] [astra_camera_node]: rgb_frame_id = 'camera_rgb_optical_frame'
[astra_camera_node-2] [INFO] [1707036651.428923673] [astra_camera_node]: depth_frame_id = 'camera_depth_optical_frame'
[astra_camera_node-2] [INFO] [1707036651.429053119] [astra_camera_node]: initDevice
[uvc_camera_node-3] [WARN] [1707036651.432208193] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[astra_camera_node-2] Warning: USB events thread - failed to set priority. This might cause loss of data...
[uvc_camera_node-3] [INFO] [1707036651.450714643] [uvc_camera_node]: Opening camera with vendor=0x2bc5, product=0x511, serial="", index=0
[component_container-1] [INFO] [1707036651.584452238] [depth_image_proc_container]: Load Library: /opt/ros/humble/lib/[libdepth_image_proc.so](http://libdepth_image_proc.so/)
[uvc_camera_node-3] unsupported descriptor subtype VS_COLORFORMAT
[uvc_camera_node-3] [INFO] [1707036651.692170197] [uvc_camera_node]: uvc mode: 640x480@30.000000 uncompressed
[uvc_camera_node-3] attempt to claim already-claimed interface 1
[component_container-1] [INFO] [1707036651.769629232] [depth_image_proc_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::ConvertMetricNode>
[component_container-1] [INFO] [1707036651.769775480] [depth_image_proc_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::CropForemostNode>
[component_container-1] [INFO] [1707036651.769802009] [depth_image_proc_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::DisparityNode>
[component_container-1] [INFO] [1707036651.769820986] [depth_image_proc_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyzNode>
[component_container-1] [INFO] [1707036651.769833787] [depth_image_proc_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyzNode>
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/point_cloud_xyz_node' in container '/depth_image_proc_container'
[astra_camera_node-2] Warning: USB events thread - failed to set priority. This might cause loss of data...
[astra_camera_node-2] [INFO] [1707036653.619307434] [astra_wrapper]: device name: Astra SV1301S_U3
[astra_camera_node-2] [INFO] [1707036653.619956683] [astra_camera_node]: depth_mode is Resolution: 640x480@30Hz Format: Depth 1mm
[astra_camera_node-2] [INFO] [1707036653.620102931] [astra_camera_node]: color mode is Resolution: 640x480@30Hz Format: RGB888
[astra_camera_node-2] [INFO] [1707036653.620180055] [astra_camera_node]: IR mode is Resolution: 640x480@30Hz Format: Gray16
[astra_camera_node-2] [ERROR] [1707036653.631410837] [astra_camera_node]: Unsupported IR video mode - Resolution: 640x480@30Hz Format: Gray16
[astra_camera_node-2] [ERROR] [1707036653.902902307] [astra_camera_node]: Unsupported depth video mode - Resolution: 640x480@30Hz Format: Depth 1mm
[astra_camera_node-2] 1
[astra_camera_node-2] [WARN] [1707036653.952561164] [astra_camera_node]: has IR sensor is 1
[astra_camera_node-2] [INFO] [1707036653.964085753] [astra_camera_node]: Starting IR stream.
[astra_camera_node-2] [WARN] [1707036653.976396558] [astra_camera_node]: has depth sensor is 1
[astra_camera_node-2] [INFO] [1707036653.981503731] [astra_camera_node]: Starting depth stream.
[... cut ...]
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[robot_state_publisher-4] [INFO] [1707036661.523141180] [rclcpp]: signal_handler(signum=2)
[INFO] [robot_state_publisher-4]: process has finished cleanly [pid 20967]
[INFO] [component_container-1]: process has finished cleanly [pid 20961]
[uvc_camera_node-3] [INFO] [1707036661.523140796] [rclcpp]: signal_handler(signum=2)
[component_container-1] [INFO] [1707036661.523170942] [rclcpp]: signal_handler(signum=2)
[INFO] [uvc_camera_node-3]: process has finished cleanly [pid 20965]

hzwsl121, 2024-02-20:
void AstraDevice::startDepthStream() {
  std::shared_ptr<openni::VideoStream> stream = getDepthVideoStream();

  if (stream) {
    stream->setMirroringEnabled(false);
    stream->start();
    stream->addNewFrameListener(depth_frame_listener.get());
    depth_video_started_ = true;
  }
}
here stream->start();
There is an error with the openni.h file.

	/**
	Starts data generation from this video stream.
	*/
	Status start()
	{
		if (!isValid())
		{
			return STATUS_ERROR;
		}
		return (Status)oniStreamStart(m_stream);
	}