orbbec/OrbbecSDK_ROS2 #8: Astra - Failed to setup devices

rafal-gorecki, 2023-10-04:
Hello,
I have two orbbec camera: "Astra" and "Astra +". Using the ROS2 repo, I configured an environment on Ubuntu 22 with ROS humble installed.
After building the package, however, I have a problem with initializing the "Astra" camera. As long as "Astra +" (ros2 launch orbbec_camera astra_adv.launch.py) starts without any problems. This is how executing ros2 launch orbbec_camera astra.launch.py ​​with only the "Astra" camera connected causes the following error:

```
Loaded node '/camera/camera' in container '/camera/camera_container'
[camera.camera]: OB_STREAM_IRsensor isn't supported by current device! -- Skipping...
[camera.camera]: OB_STREAM_COLORsensor isn't supported by current device! -- Skipping...
[camera.camera]: OB_STREAM_DEPTHsensor isn't supported by current device! -- Skipping...
[camera.camera]: Failed to setup devices: propertyManager have not been initialized
[ERROR] [camera.camera]: startDevice failed: This device has no valid sensor!
[camera.camera]: device list count 2
[camera.camera]: device list count 2
```

I will also add that I tested the devices separately (only one camera was connected at a time) and I don't know if it matters, but Astra was purchased over 1 year ago.

JGMonroy, 2023-10-19:
Same problem here!

Just one Astra camera connected, but the SDK finds 2 devices. Fails when trying to open the "default" device.

JGMonroy, 2023-10-23:
Replicated the same issue on a different robot. This time with 2 Orbbec Astra cameras. Launching the default astra.launch detects 4 devices, and fails to open the camera.

albertoezquerro, 2023-11-14:
Any updates on this? I'm facing the same issue here.