orbbec/OrbbecSDK_ROS2 #13: Failed to setup devices: Property is not supported! propertyId: 2025

NoahEleven, 2023-11-07:
我在使用Astra+适配的时候报错，主控是orinNano
‘’‘
Failed to setup devices: Property is not supported! propertyId: 2025
Failed to initialize device Invalid input, No matched video stream profile found!
’‘’

XanderWangXY, 2023-11-27:
> 我在使用Astra+适配的时候报错，主控是orinNano
> 
> ```
> Failed to setup devices: Property is not supported! propertyId: 2025
> Failed to initialize device Invalid input, No matched video stream profile found!
> ```

请问问题解决了吗

lixby03, 2023-12-12:
“Failed to initialize device Invalid input, No matched video stream profile found!”

The above error indicates that the resolution, frame rate, and data format configuration of the data stream in the current launch file is incorrect.

First, use the launch file astra_adv.launch.py to astra+. 
Then, make sure the resolution and data format for depth, color, and IR are correctly configured in the launch file.