orbbec/OrbbecSDK_ROS2 #14: Device found but not able to start [Original Astra Camera]

BruttherJOE, 2023-11-14:
Hi all, while running with original astra camera I am able to find the device but not able to start.
This is running on Ubuntu 22.04 ROS Humble.
Any help will be appreciated, attached below is the log.

```
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [8723]
[component_container-1] [INFO] [1699939706.286147973] [camera.camera_container]: Load Library: /home/joe/astra_ws/install/orbbec_camera/lib/liborbbec_camera.so
[component_container-1] [INFO] [1699939706.344701786] [camera.camera_container]: Found class: rclcpp_components::NodeFactoryTemplate<orbbec_camera::OBCameraNodeDriver>
[component_container-1] [INFO] [1699939706.344764186] [camera.camera_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<orbbec_camera::OBCameraNodeDriver>
[component_container-1] [11/14 13:28:26.348498][info][8723][Context.cpp:67] Context created with config: /home/joe/astra_ws/install/orbbec_camera/share/orbbec_camera/config/OrbbecSDKConfig_v1.0.xml
[component_container-1] [11/14 13:28:26.348508][info][8723][Context.cpp:72] Context work_dir=/home/joe/astra_ws/src/OrbbecSDK_ROS2/orbbec_camera/launch
[component_container-1] [11/14 13:28:26.352918][warning][8723][OpenNIDeviceInfo.cpp:116] New openni device matched.
[component_container-1] [11/14 13:28:26.352993][info][8723][LinuxPal.cpp:110] Create PollingDeviceWatcher!
[component_container-1] [11/14 13:28:26.353012][info][8723][DeviceManager.cpp:15] Current found device(s): (2)
[component_container-1] [11/14 13:28:26.353017][info][8723][DeviceManager.cpp:24]       - Name: BootDevice, PID: 0x0401, SN/ID: , Connection: USB2.0
[component_container-1] [11/14 13:28:26.353020][info][8723][DeviceManager.cpp:24]       - Name: Astra, PID: 0x0401, SN/ID: , Connection: USB2.0
[component_container-1] [INFO] [1699939707.354778817] [camera.camera]: device list count 2
[component_container-1] [INFO] [1699939707.354869561] [camera.camera]: Connecting to the default device
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/camera/camera' in container '/camera/camera_container'
[component_container-1] [INFO] [1699939707.356356180] [camera.camera]: OB_STREAM_IRsensor isn't supported by current device! -- Skipping...
[component_container-1] [INFO] [1699939707.356392597] [camera.camera]: OB_STREAM_COLORsensor isn't supported by current device! -- Skipping...
[component_container-1] [INFO] [1699939707.356398903] [camera.camera]: OB_STREAM_DEPTHsensor isn't supported by current device! -- Skipping...
[component_container-1] [ERROR] [1699939707.356490003] [camera.camera]: Failed to setup devices: propertyManager have not been initialized
[component_container-1] [ERROR] [1699939707.362229854] [camera.camera]: startDevice failed: This device has no valid sensor!
[component_container-1] [INFO] [1699939708.354775232] [camera.camera]: device list count 2
[component_container-1] [INFO] [1699939709.354776017] [camera.camera]: device list count 2
[component_container-1] [INFO] [1699939710.354776236] [camera.camera]: device list count 2
```

albertoezquerro, 2023-11-14:
I'm facing the same issue here. Any updates on this?

zhonghong322, 2023-12-16:
Astra cameras are not supported. We will filter out Astra cameras in future versions.

unicornuniform, 2024-01-09:
> Astra cameras are not supported. We will filter out Astra cameras in future versions.

What should someone use if they want to use an original Astra camera with ROS 2?

GabeSmythe1, 2024-06-20:
Is this still the case? The astra camera's python launch files are still being updated frequently. I'm also wondering if there is an alternative for Astra cams in ROS2.

niektuytel, 2024-07-01:
Yeah, indeed is there then a other opportunity? (i personally think this cam is still relative in development.)