orbbec/OrbbecSDK_ROS2 #16: Camera info url has no effect

huma-meng, 2023-12-07:
If a URL to a calibration file is specified, it is not taken into account. Instead of using the self-created calibration, the (presumably factory-created?) calibration stored on the camera is used. However, this calibration is completely incorrect, at least for the Astra Mini cameras that I am using.

KrzysztofDubel, 2025-07-09:
I have got the same issue still.
- Working on `v2-main` and ROS2 humble
- Tested on Astra 2 and Femto Bolt
    - Same issue on both
- Tried changing `ir_info_url` and `color_info_url`
    - Nothing changes
- Tried updating parameter at runtime 
    - `[WARN] [1752088911.857591006] [camera.camera]: Parameter ir_info_url can not be changed in runtime.`

I can also see that the CameraInfoManager pointers are not assigned to anything but `nullptr` anywhere in the code.
[image]

I ended up using `image_proc::RectifyNode` to apply the custom `camera_info` published from file by a custom node.

KrzysztofDubel, 2025-07-09:
The info urls work on `main` but I cannot set 4k resolution on femto_bolt there