orbbec/OrbbecSDK_ROS2 #19: inconsistent frame_id in ROS messages from the Gemini2

JanNogga, 2023-12-22:
Using the default Gemini2 launch file, messages from camera topics such as _/camera/color/image_raw_ or _/camera/color/camera_info_ are stamped with **header.frame_id = color_optical_frame**. On the other hand, the corresponding tf2 entry is **camera_color_optical_frame**. 
At first glance it seems like the camera message headers are perhaps ignoring the camera name, which I guess is _"camera"_ per default. Anyhow, the issue is somewhat consequential because tf2 lookups necessary to transform sensor data to other frames are failing because of this.

jian-dong, 2023-12-23:
@JanNogga Thanks for your feedback, I'll fix this next Monday.

ptzld, 2023-12-27:
Hi @jian-dong,
I am a colleague of @JanNogga and after updating our system from [05b717c](https://github.com/orbbec/OrbbecSDK_ROS2/commit/05b717c95540de066f9bed342f4c2b30bf69af47), I can confirm that the camera_info messages of the color, depth, and ir images now all point to the frame `camera_color_optical_frame`. Also, the specified launch argument `camera_name` propagates as expected through tf, topic names, and frame_ids. Thanks for your help!

closed 2023-12-27 by jian-dong