orbbec/OrbbecSDK_ROS2 #20: Different computers have different frame rates for publishing image topics

dotlikeCode, 2024-01-04:
产品 : gemini2
ros版本: humble
问题:  我在两台ubuntu 22.04的pc上，使用相同的sdk，分别启动orbbec camera，但发布的rgbd已久depth的图像帧率存在差别. 使用ros2 topic hz 查看，一台为30fps，另外一台仅仅只有20fps, 可以帮忙看一下吗
启动文件配置如下:
```xml
<launch>
    <!-- unique camera name-->
    <arg name="camera_name" default="camera"/>
    <!-- Hardware depth registration -->
    <arg name="depth_registration" default="true"/>
    <arg name="serial_number" default=""/>
    <arg name="device_num" default="1"/>
    <arg name="vendor_id" default="0x2bc5"/>
    <arg name="product_id" default=""/>
    <arg name="enable_point_cloud" default="true"/>
    <arg name="enable_colored_point_cloud" default="true"/>
    <arg name="point_cloud_qos" default="default"/>
    <arg name="connection_delay" default="100"/>
    <arg name="color_width" default="640"/>
    <arg name="color_height" default="480"/>
    <arg name="color_fps" default="5"/>
    <arg name="color_format" default="rgb"/>
    <arg name="enable_color" default="true"/>
    <arg name="flip_color" default="false"/>
    <arg name="color_qos" default="default"/>
    <arg name="color_camera_info_qos" default="default"/>
    <arg name="enable_color_auto_exposure" default="true"/>
    <arg name="depth_width" default="640"/>
    <arg name="depth_height" default="400"/>
    <arg name="depth_fps" default="30"/>
    <arg name="depth_format" default="Y14"/>
    <arg name="enable_depth" default="true"/>
    <arg name="flip_depth" default="false"/>
    <arg name="depth_qos" default="default"/>
    <arg name="depth_camera_info_qos" default="default"/>
    <arg name="depth_work_mode" default=""/>
    <arg name="ir_width" default="640"/>
    <arg name="ir_height" default="400"/>
    <arg name="ir_fps" default="30"/>
    <arg name="ir_format" default="Y8"/>
    <arg name="enable_ir" default="true"/>
    <arg name="flip_ir" default="false"/>
    <arg name="ir_qos" default="default"/>
    <arg name="ir_camera_info_qos" default="default"/>
    <arg name="enable_ir_auto_exposure" default="true"/>
    <arg name="publish_tf" default="true"/>
    <arg name="tf_publish_rate" default="10.0"/>
    <arg name="ir_info_url" default=""/>
    <arg name="color_info_url" default=""/>
    <arg name="log_level" default="info"/>
    <arg name="enable_publish_extrinsic" default="false"/>
    <arg name="enable_hardware_d2d" default="true"/>
    <arg name="enable_d2c_viewer" default="false"/>
    <arg name="enable_soft_filter" default="true"/>
    <arg name="sync_mode" default="close"/>
    <arg name="ir_trigger_signal_in_delay" default="0"/>
    <arg name="rgb_trigger_signal_in_delay" default="0"/>
    <arg name="device_trigger_signal_out_delay" default="0"/>
    <arg name="sync_signal_trigger_out" default="false"/>
    <arg name="save_image_count" default="100"/>
    <group>
....
```

jian-dong, 2024-01-09:
@dotlikeCode 你好，是同一台相机在两台电脑的的帧率输出不一样嘛，我怀疑是DDS配置区别
https://docs.ros.org/en/rolling/How-To-Guides/DDS-tuning.html

dotlikeCode, 2024-01-11:
你好，是的。我看一下dds配置

dotlikeCode, 2024-01-12:
@jian-dong  您好 ，修改DDS之后没有改善

yangyunhui1999, 2024-01-28:
我也存在这样的现象，我的现象是，两台分别都装有humble和gazebo的笔记本和台式机，在笔记本上使用gazebo相机插件仿真（1080p 60hz），发布的图像频率为设定的频率，然后运行orbbec_ros2，频率只有不到10hz(另拷贝版本也只有20hz)，反过来测试gazebo仿真的图像输出频率也降低到不到10hz，重启未见恢复。后续使用台式机测试，未见降频。

jian-dong, 2024-01-29:
@yangyunhui1999 您好，麻烦您在这里下载一个[OrbbecViewer](https://github.com/orbbec/OrbbecSDK/releases/tag/v1.9.3)看看是不是也是这个现象

yangyunhui1999, 2024-01-29:
> @yangyunhui1999 您好，麻烦您在这里下载一个[OrbbecViewer](https://github.com/orbbec/OrbbecSDK/releases/tag/v1.9.3)看看是不是也是这个现象

使用SDK没有上述问题，什么原因导致的上述现象呢，看之前的回复，如果是dds的配置问题，为何在两台未修改过dds的电脑上表现不一致

yangyunhui1999, 2024-01-29:
> > @yangyunhui1999您好，麻烦您在这里下载一个[OrbbecViewer](https://github.com/orbbec/OrbbecSDK/releases/tag/v1.9.3)看看是不是也是这个现象
> 
> 使用SDK没有出现上述问题，什么原因导致上述现象呢，看之前的回复，如果是dds的配置问题，为何在未修改过dds的电脑上表现不一致

查了一些资料，ros官方给出的解答是python和dds在序列化反序列化的性能不是很好，在使用c++订阅的时候，频率下降也存在，不过没有那么严重，因此在使用ros2 topic hz订阅频率观察到严重的频率下降，以及cli查看画面也存在卡顿。不过我在c++订阅的时候将qos设置为10,发现经常性的粘包现象