orbbec/OrbbecSDK_ROS2 #21: 3 x astra_embedded_s camera, cannot run all 3 cameras

falinuxKr, 2024-01-09:
- target: aarch64, x86_64 
- OS: ubuntu 20.04, 22.04
- ROS_DISTRO: foxy, humble
- issue: need three astra embedded S cameras [(https://shop.orbbec3d.com/Astra-Embedded-S) at a time.
       Two cameras work pretty well with pre-defined launch file, however, 3th camera does not work at all. 
       Modified launch file added. Pls review it and comment what's wrong.
      
- what I tried, 
  1) use USB 3.0 hub with external power supply
  2) PCIe to USB 3.0 extension card with separate power feeding 
  3) monitor USB bandwidth by _usbtop_ 

Thanks 

* to attach this launch file, I modified file extension from _py_ to _txt_.
[multi_camera.launch.txt](https://github.com/orbbec/OrbbecSDK_ROS2/files/13870244/multi_camera.launch.txt)

jian-dong, 2024-01-09:
Hello @falinuxKr
  what resolution are you using and how many streams are on?

falinuxKr, 2024-01-09:
> Hello @falinuxKr what resolution are you using and how many streams are on?

Hello @jian-dong,
By default, all cameras-RGB, Depth, IR- have 640x480 30fps by default.

jian-dong, 2024-01-10:
Hello @falinuxKr 
This resolution setting exceeds the maximum bandwidth of USB 3.0; you can lower the resolution and frame rate and try again.

falinuxKr, 2024-01-12:
@jian-dong 

I changed Depth to 320x200 @5, and IR to 320x200@5. RGB as it is.
According to` list_camera_profile_mode_mode`, RGB does not have lower mode than 640x480@30. 
But failed to see all 3 cameras. Depth and IR of one camera show very low outputs(fps) instead of other 2 cameras.

zhonghong322, 2024-01-15:
Please modify the buffer size of the kernel and test it with the following command, making sure to run it with root permissions.

echo 128 > /sys/module/usbcore/parameters/usbfs_memory_mb     
cat /sys/module/usbcore/parameters/usbfs_memory_mb

After modifying the kernel buffer size, we tested it with three devices and it worked fine.
[image]

falinuxKr, 2024-01-15:
@zhonghong322
VERY thanks. You saved lots of time.

closed 2024-01-24 by jian-dong