orbbec/OrbbecSDK_ROS2 #23: Does it support starting from an IP address?

ygao9795, 2024-01-19:
Hello,

I am curious to know if it supports starting the camera using an IP address. My current requirement involves connecting to the camera over the network.

Thank you for your assistance!

zhonghong322, 2024-01-30:
What device are you using? The Femto Mega supports network .

ygao9795, 2024-02-05:
我的相机型号是Femto Mega，我找到了网络启动参数：`enumerate_net_device`

我在femto_mega.launch中做了修改：`DeclareLaunchArgument('enumerate_net_device', default_value='true')`，但是还是找不到设备。

我在OrbbecViewer中可以正常通过网络打开相机，不知道在ros2中是否需要有其它的设置。

zhonghong322, 2024-02-06:
1、首先确定下Femto Mega 和 PC 能够ping通。 
2. 升级Femto Mega的固件 到1.2.7或 以上版本，这个固件版本开始才又设备发现功能，固件升级链接如下：  
https://github.com/orbbec/OrbbecFirmware/releases/tag/Femto-Mega-Firmware
4. 排查下PC 是不是有多个网络，如果有多个，关掉其它的网络试下。

ygao9795, 2024-02-06:
谢谢您的及时回复。
1. 可以正常ping通
2. 固件升级到了1.2.8
3. 确实是因为多个网络的原因，我把wifi关掉以后，可以成功启动了

再次感谢！

closed 2024-03-03 by ygao9795