orbbec/OrbbecSDK_ROS2 #27: How to adjust depth parameter spec

ilileun, 2024-03-28:
[image]
        DeclareLaunchArgument('color_width', default_value='1280'),
        DeclareLaunchArgument('color_height', default_value='720'),
        DeclareLaunchArgument('color_fps', default_value='30'),
        DeclareLaunchArgument('color_format', default_value='MJPG'),
        DeclareLaunchArgument('enable_color', default_value='true'),
        DeclareLaunchArgument('flip_color', default_value='false'),
        DeclareLaunchArgument('color_qos', default_value='default'),
        DeclareLaunchArgument('color_camera_info_qos', default_value='default'),
        DeclareLaunchArgument('enable_color_auto_exposure', default_value='true'),
        DeclareLaunchArgument('depth_width', default_value='1024'),
        DeclareLaunchArgument('depth_height', default_value='1024'),
        DeclareLaunchArgument('depth_fps', default_value='15'),
        

>> Depth image is not show in the rviz2. 
>> How to know all  Valid options???

3Ddaiwei, 2024-12-05:
You can try to configure the effective depth image resolution. Refer to the product specification for setting the supported resolution.

3Ddaiwei, 2024-12-05:
It is recommended to try using "ros2 topic hz /camera/depth/image_raw" to check whether the depth frame rate is output normally, and then check whether the topic configuration on rviz is correct.