orbbec/OrbbecSDK_ROS2 #32: ERROR  FAILED to get depth: Stream Profil

tottichopper, 2024-05-14:
Hi,
Im on humble-dist. and installed the openni sdk to use the the orbbecsdk_ros2 wrapper for my Astra S.
the Following error is trigger from launch. i tried all launch files only dabai.dw.launch is wrkin but no image etc ??
Error code from the launch:

[INFO] [launch]: All log files can be found below /home/userdistlab/.ros/log/2024-05-14-13-14-11-497996-LaptopDistLab-88252
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [88264]
[component_container-1] [INFO] [1715685251.812279171] [camera.camera_container]: Load Library: /home/userdistlab/astra_ws/install/orbbec_camera/lib/liborbbec_camera.so
[component_container-1] [INFO] [1715685251.875121191] [camera.camera_container]: Found class: rclcpp_components::NodeFactoryTemplate<orbbec_camera::OBCameraNodeDriver>
[component_container-1] [INFO] [1715685251.875580085] [camera.camera_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<orbbec_camera::OBCameraNodeDriver>
[component_container-1] [05/14 13:14:11.879370][info][88264][Context.cpp:69] Context created with config: /home/userdistlab/astra_ws/install/orbbec_camera/share/orbbec_camera/config/OrbbecSDKConfig_v1.0.xml
[component_container-1] [05/14 13:14:11.879725][info][88264][Context.cpp:74] Context work_dir=/home/userdistlab
[component_container-1] [05/14 13:14:11.879893][info][88264][Context.cpp:77] 	- SDK version: 1.9.7
[component_container-1] [05/14 13:14:11.880047][info][88264][Context.cpp:78] 	- SDK stage version: main
[component_container-1] [05/14 13:14:11.880190][info][88264][Context.cpp:82] get config EnumerateNetDevice:false
[component_container-1] [05/14 13:14:11.880330][info][88264][LinuxPal.cpp:39] createObPal: create LinuxPal!
[component_container-1] [05/14 13:14:12.000877][warning][88264][OpenNIDeviceInfo.cpp:122] New openni device matched.
[component_container-1] [05/14 13:14:12.000927][info][88264][LinuxPal.cpp:116] Create PollingDeviceWatcher!
[component_container-1] [05/14 13:14:12.000938][info][88264][DeviceManager.cpp:15] Current found device(s): (1)
[component_container-1] [05/14 13:14:12.000940][info][88264][DeviceManager.cpp:24] 	- Name: Astra S, PID: 0x0402, SN/ID: , Connection: USB2.0
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/camera/camera' in container '/camera/camera_container'
[component_container-1] [INFO] [1715685252.101438059] [camera.camera]: Connecting to the default device
[component_container-1] [INFO] [1715685252.119984391] [camera.camera]: OpenNI device detected.
[component_container-1] [INFO] [1715685252.128506315] [camera.camera]:  stream depth is enabled - width: 640, height: 480, fps: 30, Format: OB_FORMAT_Y11
[component_container-1] [ERROR] [1715685252.128577655] [camera.camera]: Failed to get ir << profile: Invalid input, No matched video stream profile found!
[component_container-1] [ERROR] [1715685252.128584285] [camera.camera]: Stream: OB_STREAM_IR, Stream Index: 0, Width: 640, Height: 400, FPS: 30, Format: OB_FORMAT_Y10
[ERROR] [component_container-1]: process has died [pid 88264, exit code 255, cmd '/opt/ros/humble/lib/rclcpp_components/component_container --ros-args -r __node:=camera_container -r __ns:=/camera'].

maybe im passing the wrong parameters or i got the wrong SDK IDK

jian-dong, 2024-06-11:
Astra S is not support by this SDK

tottichopper, 2024-06-11:
and wich driver software support the astra s?

tottichopper, 2024-06-11:
> Astra S is not support by this SDK

and wich driver software support the astra s?