orbbec/OrbbecSDK_ROS2 #38: Gemini 335系列的imu参数是多少呀，accelerometer_noise_density、accelerometer_random_walk、gyroscope_noise_density、gyroscope_random_walk

lturing, 2024-06-05:


jian-dong, 2024-06-05:
在 imu info 里面可以拿到

```bash
ros2 topic  echo /camera/gyro/imu_info
ros2 topic  echo /camera/accel/imu_info
```

lturing, 2024-06-05:
感谢回复。
运行上述命令，这几个数据都为0，是正常的吗
ros2 topic  echo /camera/gyro/imu_info 部分输出如下：
```
header:
  stamp:
    sec: 1717589178
    nanosec: 502571008
  frame_id: camera_gyro_optical_frame
noise_density: 0.0
random_walk: 0.0
reference_temperature: 0.0
bias:
- 0.0
- 0.0
- 0.0
gravity:
- 0.0
- 0.0
- 0.0
scale_misalignment:
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
temperature_slope:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
---

```

ros2 topic  echo /camera/accel/imu_info

```
header:
  stamp:
    sec: 1717588773
    nanosec: 860081697
  frame_id: camera_accel_optical_frame
noise_density: 0.0
random_walk: 0.0
reference_temperature: 0.0
bias:
- 0.0
- 0.0
- 0.0
gravity:
- 0.0
- 0.0
- 0.0
scale_misalignment:
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
temperature_slope:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
---
```

closed 2024-06-08 by lturing