orbbec/OrbbecSDK_ROS2 #40: Error Building orbbec_description (RVIZ)

Edwardius, 2024-06-11:
```
CMake Error at cmake_install.cmake:41 (file):
  file INSTALL cannot find
  "/home/USERNAME/ros2_ws/src/OrbbecSDK_ROS2/orbbec_description/rviz": No
  such file or directory.

--- stderr: orbbec_description                                                                                                   
CMake Error at cmake_install.cmake:41 (file):
  file INSTALL cannot find
  "/home/USERNAME/ros2_ws/src/OrbbecSDK_ROS2/orbbec_description/rviz": No
  such file or directory.

---
```
This seems to have been caused by the most recent commit 10148df62e15bd51a1993a8bf2c4c03428401c9e

Kashery, 2024-06-12:
You can fix it in a "work around" fashion by removing line 10 at OrbbecSDK_ROS2/orbbec_description/CMakeLists.txt

So CMakeLists.txt should look something like that:
```cmake
cmake_minimum_required(VERSION 3.5)
project(orbbec_description)

find_package(ament_cmake REQUIRED)

# Install files
install(DIRECTORY 
        launch 
        meshes 
        urdf
        DESTINATION share/${PROJECT_NAME})

ament_package()
```
keep in mind that this just ignores the rviz directory and will simply ommit it, im not convinced that this is an acutall fix.

jian-dong, 2024-06-22:
@Kashery Thank you for your contribution

closed 2024-06-22 by jian-dong