orbbec/OrbbecSDK_ROS2 #47: rockchip 编译报错动态链接库

suse-chen, 2024-07-01:
/usr/bin/ld:/home/cat/ros2_ws/src/OrbbecSDK_ROS2-main/orbbec_camera/SDK/lib/arm64/libOrbbecSDK.so: file format not recognized; treating as linker script
/usr/bin/ld:/home/cat/ros2_ws/src/OrbbecSDK_ROS2-main/orbbec_camera/SDK/lib/arm64/libOrbbecSDK.so:0: syntax error
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/orbbec_camera.dir/build.make:485: liborbbec_camera.so] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:145: CMakeFiles/orbbec_camera.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
--- stderr: orbbec_camera                                        
/usr/bin/ld:/home/cat/ros2_ws/src/OrbbecSDK_ROS2-main/orbbec_camera/SDK/lib/arm64/libOrbbecSDK.so: file format not recognized; treating as linker script
/usr/bin/ld:/home/cat/ros2_ws/src/OrbbecSDK_ROS2-main/orbbec_camera/SDK/lib/arm64/libOrbbecSDK.so:0: syntax error
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/orbbec_camera.dir/build.make:485: liborbbec_camera.so] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:145: CMakeFiles/orbbec_camera.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< orbbec_camera [1min 3s, exited with code 2]

jian-dong, 2024-07-03:
是不是在windows上解压过了，直接把在rk上解压试一下

mfjiang, 2026-02-25:
libOrbbecSDK.so 写着 libOrbbecSDK.so.2
libOrbbecSDK.so.2 写着 libOrbbecSDK.so.2.7.6

file ./binary/rootfs/opt/ros2_ws/src/OrbbecSDK_ROS2-2-main/orbbec_camera/SDK/lib/
arm64/libOrbbecSDK.so.2.7.6
./binary/rootfs/opt/ros2_ws/src/OrbbecSDK_ROS2-2-main/orbbec_camera/SDK/lib/arm64/libOrbbecSDK.so.2.7.6: ELF 64-bit LSB shared object, ARM aarch64, version 1 (GNU/Linux), dynamically linked, BuildID[sha1]=370a3f178e4b090534325db17241c5ced8b329c0, not stripped

mfjiang, 2026-02-25:
fix it:
LIBDIR="/opt/ros2_ws/src/OrbbecSDK_ROS2-2-main/orbbec_camera/SDK/lib/arm64"
if [ ! -L "${LIBDIR}/libOrbbecSDK.so" ]; then
    echo "🔧 修复 libOrbbecSDK.so 符号链接"
    rm -f "${LIBDIR}/libOrbbecSDK.so"
    ln -s libOrbbecSDK.so.2 "${LIBDIR}/libOrbbecSDK.so"
fi
if [ ! -L "${LIBDIR}/libOrbbecSDK.so.2" ]; then
    echo "🔧 修复 libOrbbecSDK.so.2 符号链接"
    rm -f "${LIBDIR}/libOrbbecSDK.so.2"
    ln -s libOrbbecSDK.so.2.7.6 "${LIBDIR}/libOrbbecSDK.so.2"
fi