orbbec/OrbbecSDK_ROS2 #52: Coordinate System Orientation issue with camera_depth_optical_frame and camera_color_optical_frame in Femto Bolt

Danilrivero, 2024-07-08:
# General information

Ubuntu: 22.04 LTS
ROS: Humble
Driver version: 1.58

## Explanation

This issue is directly related to my comment in this issue: https://github.com/orbbec/OrbbecSDK_ROS2/issues/44#issuecomment-2196523231.

You have merged the proposed changes here https://github.com/orbbec/OrbbecSDK_ROS2/issues/44#issuecomment-2196132794 but the actual orientation of the axis may need to be different in order to follow the real model.

## Expected output

The Femto Bolt camera is a continuation of the Azure Kinect that has the following frame orientation in it's driver:

[image]

As you can see the axis orientation follows the order I proposed in my comment and should be kept the same in this driver due to the same HW orientation in both cameras.

1. The _camera_depth_optical_frame_ contains a rotation without translation related to _camera_base_link_ which **remains unaffected**.

[image]

2. The relation between _camera_color_optical_frame_ and _camera_base_link_ is a **translation**.

[image]

## Actual output

The frame orientation is wrong in relation to the real HW orientation:

[image]

---

These changes are not crucial since the actual pointcloud does reflect the correct transformation but in order to have a usable TF tree for representation and visualisation it could be helpful.

- Ideally the driver shall have a URDF file to provide a visualisation of the TF tree for a robot such as proposed here: https://github.com/orbbec/OrbbecSDK_ROS2/issues/51
- This visualisation should be matched with a proper camera mesh with it's frame correctly aligned as proposed here: https://github.com/orbbec/OrbbecHardware/issues/2

Opening this issue to let comment on the differences between this driver and the Azure Kinect driver from Microsoft from which many users probably come.

jian-dong, 2024-07-08:
I didn't quite understand your question, perhaps [this link](https://www.orbbec.com/documentation/femto-bolt-coordinate-systems/)could be helpful.

Danilrivero, 2024-07-08:
> I didn't quite understand your question, perhaps [this link](https://www.orbbec.com/documentation/femto-bolt-coordinate-systems/) could be helpful.

As shown here the _camera_base_link_ and _depth_camera_optical_frame_ should **not** coincide and have a rotation of about 6 degrees between them:

[image]

If we follow the rules from said documentation the resulting axis would look like this:

[image]

**In reality** _camera_base_link_ and _depth_camera_optical_frame_ **are aligned** when you check the TF tree in RVIZ2:

[image]

---

| How the frames should look like | How the frames look like right now | 
| -------- | -------- | 
|  [image] |  [image]  | 

To be clear, the driver works fine and the resulting pointcloud given by it shows no issue. I opened the issue to understand why the frames do not match the drawings. If the user needs to transform the reference frame of the pointcloud which is color_camera_link to any other, since there is a rotation between this link and the camera_link, a wrong 6 degree rotation would be applied to it instead of only a translation.

[image]

Thanks in advance!

jian-dong, 2024-07-08:
I understand what you mean now. You're saying that the `camera_link's` x-coordinate should be facing forward, not coinciding with the `depth_frame.` I appreciate your help.

jian-dong, 2024-07-08:
@Danilrivero I think It should be like this, it's correct?

[image]

Danilrivero, 2024-07-08:
> @Danilrivero I think It should be like this, it's correct?
> 
> 
> 
> [image]
> 
> 
> 
> 

That is exactly what I meant, yes. It would be on par with how the hardware looks to be designed to properly transform between different frames.

Looks good to me! Thanks

Danilrivero, 2024-07-09:
Seems like the issue is being solved, will be closing this, thanks!

closed 2024-07-09 by Danilrivero