orbbec/OrbbecSDK_ROS2 #53: Depth Image Not Publishing on ROS2 with Femto Mega Camera

hamamabrouk, 2024-07-12:
Hello,

I am encountering an issue with the **Orbbec Femto Mega** camera while using ROS2 Humble. Despite configuring the camera node and setting the necessary parameters, the depth image is not being published on the /camera/depth/image_raw topic. Below are the details of my setup and the steps I've taken to troubleshoot the issue:

**1. System Information:**
-         Camera Model: Orbbec Femto Mega
-         ROS2 Distro: Humble
-         Firmware Version: 1.2.7
-         SDK Version: v1.10.9-20240705-3a55569
-         Connection: USB 3.2 and Ethernet (tested both)

**2. Launch File Configuration:**
I have used the launch file provided in your repository.

**3. Observations:**  
- The IR and RGB images are being published successfully.
- The depth image, however, is not being published on the /camera/depth/image_raw topic.
- The point cloud data, which depends on the depth image, is being published, indicating that the depth information is being processed but not published.

**4. Additional Tests:**
- I also tested the setup using an Ethernet connection, but the issue persists.
- Verified the topics using ros2 topic list and ros2 topic echo /camera/depth/image_raw, confirming that the data in depth topic is missing.

**5. Parameter List:**
-     I retrieved all parameters using ros2 param list /orbbec_camera_node and ros2 param get commands, ensuring they are correctly set.

Despite these steps, the depth image is still not being published. Any guidance or suggestions on how to resolve this issue would be greatly appreciated.

Thank you for your assistance.

zhonghong322, 2024-07-24:
Our engineer has fixed this bug. Is your issue resolved now?

Shreyaskar05, 2025-12-10:
@zhonghong322 could you please look into it and help us