orbbec/OrbbecSDK_ROS2 #55: Incorrect point cloud scale with Gemini 2L

danpprince, 2024-08-12:
The latest commit on `main` seems to give incorrect scale to point clouds with the Gemini 2L and the provided launch file, with the reported depth being about 1/4th the expected value. 

Using the latest commit on `main` at the time of writing:
```
$ git rev-parse HEAD
455f00294373766b1f1ff3eb4173f55147cf0da2
```
Compiled with `colcon build` and running the provided launch file:
```
$ ros2 launch orbbec_camera gemini2L.launch.py 
```
I see the point cloud in rviz in the screenshot below. Measuring with the rviz focal point, the wall is about 17 cm away.
[image]

Using the latest tag `v1.5.8`, I created the screenshot below. Measuring with the focal point in rviz, I see the wall is about 70 cm away.

[image]

The expected depth to the wall is 70 cm based on a tape measure.

I used the following Dockerfile to run this with ROS 2 Jazzy and Ubuntu Noble:
```
# Collect the arguments for the base ros image
ARG ROS_VERSION=jazzy
ARG UBUNTU_VERSION=noble

FROM ros:${ROS_VERSION}-ros-base-${UBUNTU_VERSION}

RUN apt-get update && apt-get upgrade -y

RUN apt update && apt install -y \
    libgflags-dev \
    nlohmann-json3-dev  \
    ros-${ROS_DISTRO}-desktop \
    ros-${ROS_DISTRO}-image-transport \
    ros-${ROS_DISTRO}-image-publisher \
    ros-${ROS_DISTRO}-camera-info-manager \
    ros-${ROS_DISTRO}-diagnostic-updater \
    ros-${ROS_DISTRO}-diagnostic-msgs

# Setup .bashrc
RUN echo "source /opt/ros/${ROS_DISTRO}/setup.bash" >> /home/${username}/.bashrc
RUN echo "export ROS_WORKSPACE=/home/${username}/project" >> /home/${username}/.bashrc

ENTRYPOINT ["/bin/bash"]
```

Porridge0216, 2024-09-27:
The same problem with depth image.

danpprince, 2024-09-27:
I did hear from Orbbec support, and they recommended to use the latest tag v1.5.8 instead of the default branch main. That does resolve the depth issue for me.

jian-dong, 2024-09-29:
> The same problem with depth image.

Hi @danpprince,

Are you using the main branch? I have tested it currently, and the point cloud distance matches the expected results.