orbbec/OrbbecSDK_ROS2 #56: How to Change the ROI of Automatic Exposure in OrbbecSDK_ROS2

wataru-okumura, 2024-08-16:
Thanks again.
I use OrbbecSDK_ROS2 all the time. The camera I use is a Femto Bolt.
I would like to know how to change the ROI of the automatic exposure via OrbbecSDK_ROS2.
I can turn on/off the auto exposure by switching the value of the ros parameters enable_color_auto_exposure and enable_ir_auto_exposure, but I have not been able to find the ros parameter to change the ROI setting.
I would like to know if there is a way to change the ROI of the auto exposure in OrbbecSDK_ROS2 at this time.
I would be very happy if you could define those values in ros parameter if possible.

Thank you in advance.

jian-dong, 2024-08-19:
Femto Bolt doesn't support ROI AE.

wataru-okumura, 2024-08-19:
I see. I guess I was mistaken.
Thank you for answering.

closed 2024-08-19 by wataru-okumura