orbbec/OrbbecSDK_ROS2 #58: Parameter to specify the frame_id used for all published images and point clouds.

Danilrivero, 2024-08-23:
# Suggestion

Hello,

When using the camera with a robot it is useful to calibrate it in order to obtain a **calibrated frame** that removes any error created by the differences between the real hardware and the TF being published.

## Proposal

In other drivers such as the [ROS driver for the ZIVID cameras](https://github.com/zivid/zivid-ros) or the Intel cameras, you have the ability to change the _frame_id_ to which the pointcloud is referred. 

[image]
[image]

It simply changes the **field of header/frame_id of the topic** that publishes the topic and images to a proper frame that is calibrated for the given robot or to any frame really that the user wants to refer everything to.

Warm regards.

Danilrivero, 2024-09-12:
Hello, is there any bandwith available for this? If time is an issue I could go into the code and recreate the changes made by the ZIVID driver to equally implement it.

Thanks in advance.

jian-dong, 2024-09-12:
@Danilrivero 
Apologies, I’ve been quite busy lately. I’ll add one later today and check if it meets your expectations.

jian-dong, 2024-09-12:
Hi @Danilrivero 
I checked the camera param by` ros2 param list`, and I see all frames are configurable.

```
ros2 param list                                        
/camera/camera:
  .color.image_raw.format
  .color.image_raw.jpeg_quality
  .color.image_raw.png_level
  .color.image_raw.tiff.res_unit
  .color.image_raw.tiff.xdpi
  .color.image_raw.tiff.ydpi
  .depth.image_raw.format
  .depth.image_raw.jpeg_quality
  .depth.image_raw.png_level
  .depth.image_raw.tiff.res_unit
  .depth.image_raw.tiff.xdpi
  .depth.image_raw.tiff.ydpi
  accel_optical_frame_id
  accel_qos
  accel_range
  accel_rate
  align_mode
  align_target_stream
  angular_vel_cov
  camera_accel_frame_id
  camera_color_frame_id
  camera_depth_frame_id
  camera_gyro_frame_id
  camera_ir_frame_id
  camera_left_ir_frame_id
  camera_name
  camera_right_ir_frame_id
  color.image_raw.enable_pub_plugins
  color_camera_info_qos
  color_delay_us
  color_exposure
  color_format
  color_fps
  color_gain
  color_height
  color_info_url
  color_optical_frame_id
  color_qos
  color_white_balance
  color_width
  connection_delay
  decimation_filter_scale_
  depth.image_raw.enable_pub_plugins
  depth_camera_info_qos
  depth_delay_us
  depth_filter_config
  depth_format
  depth_fps
  depth_height
  depth_optical_frame_id
  depth_precision
  depth_qos
  depth_registration
  depth_width
  depth_work_mode
  device_num
  device_preset
  diagnostic_period
  diagnostic_updater.period
  enable_3d_reconstruction_mode
  enable_accel
  enable_color
  enable_color_auto_exposure
  enable_color_auto_white_balance
  enable_color_undistortion
  enable_colored_point_cloud
  enable_d2c_viewer
  enable_decimation_filter
  enable_depth
  enable_depth_scale
  enable_frame_sync
  enable_gyro
  enable_hardware_d2d
  enable_hdr_merge
  enable_heartbeat
  enable_hole_filling_filter
  enable_ir
  enable_ir_auto_exposure
  enable_ir_long_exposure
  enable_laser
  enable_ldp
  enable_left_ir
  enable_noise_removal_filter
  enable_point_cloud
  enable_right_ir
  enable_sequence_id_filter
  enable_soft_filter
  enable_spatial_filter
  enable_sync_host_time
  enable_sync_output_accel_gyro
  enable_temporal_filter
  enable_threshold_filter
  enumerate_net_device
  flip_color
  flip_depth
  flip_ir
  flip_left_ir
  flip_right_ir
  frames_per_trigger
  gyro_optical_frame_id
  gyro_qos
  gyro_range
  gyro_rate
  hdr_merge_exposure_1
  hdr_merge_exposure_2
  hdr_merge_gain_1
  hdr_merge_gain_2
  hole_filling_filter_mode
  ir_camera_info_qos
  ir_exposure
  ir_format
  ir_fps
  ir_gain
  ir_height
  ir_info_url
  ir_optical_frame_id
  ir_qos
  ir_width
  laser_energy_level
  laser_on_off_mode
  left_ir_camera_info_qos
  left_ir_format
  left_ir_fps
  left_ir_height
  left_ir_optical_frame_id
  left_ir_qos
  left_ir_width
  linear_accel_cov
  log_level
  max_depth_limit
  max_save_images_count
  min_depth_limit
  net_device_ip
  net_device_port
  noise_removal_filter_max_size
  noise_removal_filter_min_diff
  ordered_pc
  point_cloud_qos
  publish_tf
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  qos_overrides./tf.publisher.depth
  qos_overrides./tf.publisher.durability
  qos_overrides./tf.publisher.history
  qos_overrides./tf.publisher.reliability
  qos_overrides./tf_static.publisher.depth
  qos_overrides./tf_static.publisher.history
  qos_overrides./tf_static.publisher.reliability
  retry_on_usb3_detection_failure
  right_ir_camera_info_qos
  right_ir_format
  right_ir_fps
  right_ir_height
  right_ir_optical_frame_id
  right_ir_qos
  right_ir_width
  sequence_id_filter_id
  serial_number
  soft_filter_max_diff
  soft_filter_speckle_size
  software_trigger_period
  spatial_filter_alpha
  spatial_filter_diff_threshold
  spatial_filter_magnitude
  spatial_filter_radius
  sync_mode
  temporal_filter_diff_threshold
  temporal_filter_weight
  tf_publish_rate
  threshold_filter_max
  threshold_filter_min
  time_domain
  trigger2image_delay_us
  trigger_out_delay_us
  trigger_out_enabled
  usb_port
  use_sim_time
```

jian-dong, 2024-09-12:
If you want to change the `color_optical_frame_id`, use the following line:
```python
DeclareLaunchArgument('color_optical_frame_id', 'my_frame_id')
```
In the launch file. This will set the `color_optical_frame_id` to `'my_frame_id'`.

Danilrivero, 2024-09-12:
Hello, would it be possible to apply the same structure to the pointcloud? 

The generated pointcloud from depth and color images is correct. I simply need it's ```frame_id``` field to be changed like you have proposed from the original ```'camera_color_optical_frame'``` to my new frame_id ```'calibrated_camera_color_optical_frame'```. It seems like there is no such parameter available for the pointcloud, only for the color and depth images.

Thanks for your swift response and help with the issue as well

jian-dong, 2024-09-13:
@Danilrivero 
I got your meaning. Can I add point cloud frame_id but not add it to the tf tree?

Danilrivero, 2024-09-13:
> @Danilrivero I got your meaning. Can I add point cloud frame_id but not add it to the tf tree?

Exactly, when it gets published I can change the ```'frame_id'``` value only, this way, when I do a _tf transformation_ (because the new  ```'frame_id'``` I am using is being published and exists in the _tf tree_) I can properly do it without having to do the hassle of changing it myself.

In short, changing the value of std_msgs/msg/Header.msg/string frame_id for **each** pointcloud being published at runtime would be sufficient. Checking your commit it seems to be exactly what I was looking for, this will be helpful for many. I will test it as soon as possible and close the issue if solved.

I would like to thank you again for your swift response and help, cheers.

closed 2024-09-18 by jian-dong