orbbec/OrbbecSDK_ROS2 #62: Synchronisation

Withey958, 2024-09-27:
I am looking to run 3 Astra 2 cameras at the same time synchronised, using the sync hub dev, daisy chain. I am looking for some advise regarding the launch file.

I am basing the launch off `multi_camera_synced.launch.py` and updating the launch parameters eg:

```
    front_camera = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(
            os.path.join(launch_file_dir, 'gemini_330_series.launch.py')
        ),
        launch_arguments={
            'camera_name': 'front_camera',
            'usb_port': '2-1.1',
            'device_num': '4',
            'sync_mode': 'software_triggering',
            'config_file_path': config_file_path,
        }.items()
    )
```

For sync_mode which is the correct selection for my setup? Also within launch parameters if I wish to change the `depth_width` and `depth_height` should I be adding to this file or editing the default values in `astra2.launch.py`.

jian-dong, 2024-09-29:
Hi @Withey958,

For the `sync_mode` setting, please refer to this document: [Set Up Cameras for External Synchronization](https://www.orbbec.com/docs/set-up-cameras-for-external-synchronization_v1-2/). The appropriate setting may vary depending on the specific application scenario.

Regarding `depth_width` and `depth_height,` the parameters in the [camera_params.yaml](https://github.com/orbbec/OrbbecSDK_ROS2/blob/main/orbbec_camera/config/camera_params.yaml) file will override the ones in the `astra2.launch.py`.

Withey958, 2024-11-05:
Sorry I have been away for a while can I confirm in this instance I would be using:

_'sync_mode': 'primary'_
_'sync_mode': 'secondary_synced'_

I found the list of details for the astra 2:

[image]

But I can't find the input string options for 'sync_mode' in the ROS2 library, is this case sensitive?

Withey958, 2024-11-11:
I am also getting a bit confused by the documentation. Should I be running the transform as described here? 

https://github.com/orbbec/OrbbecSDK_ROS2/blob/main/docs/multi_camera.MD

Do I then also have to use these transforms to synchronize the time stamps?

Finally when I query the depth precision parameter I get no string back, is it possible to change the depth precision on the Astra 2 using the ROS2 wrapper?