orbbec/OrbbecSDK_ROS2 #63: Can you add a syncrhonized RGB-D topic?

JunweiLiang, 2024-10-09:
Hi,

Currently the color image and depth image are published in separate topics. Can you add a RGB-D topic, where the color and depth images are synchronized and sent in a single message?

You already have RGBD.msg [here](https://github.com/orbbec/OrbbecSDK_ROS2/blob/main/orbbec_camera_msgs/msg/RGBD.msg).
Realsense provides this RGB-D topic as shown [here](https://github.com/IntelRealSense/realsense-ros?tab=readme-ov-file#rgbd-topic).

Currently, you need an ApproximateTimeSynchronizer to achieve this as in [here](https://github.com/orbbec/OrbbecSDK_ROS2/issues/37#issuecomment-2149734469).

Thanks!
Junwei

jian-dong, 2024-10-12:
Hi, @JunweiLiang. Thank you for your feedback. This feature will be added to the next release, possibly next week.

Danilrivero, 2025-02-12:
> Hi, [@JunweiLiang](https://github.com/JunweiLiang). Thank you for your feedback. This feature will be added to the next release, possibly next week.

Hello, has this been implemented yet on any branch? Is there any flag available to enable this functionality? Tested both main and v2main with the Femto Bolt and it does not seem to be published.

Thanks!

NathanSchomer, 2025-11-21:
@Danilrivero did you get an answer to this question? I'm currently dealing with the same issue. Thanks!