orbbec/OrbbecSDK_ROS2 #64: Gemini 336 Depth Map Resolution Alternates

FabienDanieau, 2024-10-10:
Hi,

I have a Gemini 336 with the 1.3.25 firmware. I've observed something strange with the RO2 driver (running under Humble).

The resolution of the published depth map (/camera/depth/image_raw) alternates between 1280x720 and 848x480. If I specify depth_width:=1280, the resolution is stable, but the depth image is still "flickering." See the screencast extracted from Rviz.

[image]

Any idea what could cause this? I don't see that with the Orbbec viewer.
Thanks

jian-dong, 2024-10-12:
Hi @FabienDanieau 
Do you use the latest version ROS2 SDK?

FabienDanieau, 2024-10-14:
> Hi @FabienDanieau Do you use the latest version ROS2 SDK?

Yes 1.5.8

jian-dong, 2024-10-14:
@FabienDanieau can you try the main branch. I believe the main branch is fixed.

FabienDanieau, 2024-10-17:
Indeed it works perfectly with the main branch. Thanks!

closed 2024-10-17 by FabienDanieau