orbbec/OrbbecSDK_ROS2 #65: Empty camera info in higher resolutions (astra pro plus)

tfoldi, 2024-10-13:
Everything works fine if I am running the nodes in 640x480 resolution:

```
ros2 launch orbbec_camera astra.launch.py enable_depth:=false enable_ir:=false  color_fps:=30 
```

and camera info looks good:

```yaml
$ ros2 topic echo /camera/color/camera_info --once
header:
  stamp:
    sec: 1728817915
    nanosec: 232680960
  frame_id: camera_color_optical_frame
height: 480
width: 640
distortion_model: rational_polynomial
d:
- 0.10524643957614899
- -0.14457714557647705
- -0.0004056328325532377
- 0.0021555349230766296
- -0.09454253315925598
- 0.0
- 0.0
- 0.0
k:
- 544.0361328125
- 0.0
- 315.89178466796875
- 0.0
- 544.0361328125
- 235.51173400878906
- 0.0
- 0.0
- 1.0
r:
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
p:
- 544.0361328125
- 0.0
- 315.89178466796875
- 0.0
- 0.0
- 544.0361328125
- 235.51173400878906
- 0.0
- 0.0
- 0.0
- 1.0
- 0.0
binning_x: 0
binning_y: 0
roi:
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: false
---
```   

When I am switching to larger resolution: 

```
ros2 launch orbbec_camera astra.launch.py enable_depth:=false enable_ir:=false  color_fps:=30 color_width:=1920 color_height:=1080
```

Then camera_info becomes empty:

```
$ ros2 topic echo /camera/color/camera_info --once
header:
  stamp:
    sec: 1728817994
    nanosec: 588848128
  frame_id: camera_color_optical_frame
height: 1080
width: 1920
distortion_model: rational_polynomial
d:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
k:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 1.0
r:
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
p:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 1.0
- 0.0
binning_x: 0
binning_y: 0
roi:
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: false
---
```

Any idea how can this be fixed?

jian-dong, 2024-10-14:
Hi @tfoldi 
The color high resolution of Astra Pro Plus is indeed not calibrated.

tfoldi, 2024-10-15:
then how can I add my own camera calibration data? if you are open I can draft a PR where one can specify a parameter pointing to a calibration file.