orbbec/OrbbecSDK_ROS2 #66: Encountered `usbEnumerator create UnpackDevicePort failed!` error

bd-fauna, 2024-10-14:
When running this command:

```bash
ros2 run orbbec_camera list_devices_node
```

I get the following error:

```bash
[10/14 15:42:32.536321][info][593222][Context.cpp:68] Context created with config: default config!
[10/14 15:42:32.536331][info][593222][Context.cpp:73] Work directory=/workspaces/isaac_ros-dev, SDK version=v1.10.15-20241012-bb4e350
[10/14 15:42:32.536343][info][593222][LinuxPal.cpp:32] createObPal: create LinuxPal!
[10/14 15:42:32.548750][info][593222][LinuxPal.cpp:166] Create PollingDeviceWatcher!
[10/14 15:42:32.548765][info][593222][DeviceManager.cpp:15] Current found device(s): (1)
[10/14 15:42:32.548899][info][593222][DeviceManager.cpp:24] 	- Name: Femto Bolt, PID: 0x066b, SN/ID: , Connection: USB3.1
[ERROR] [1728934953.049575478] [list_device_node]: usbEnumerator create UnpackDevicePort failed!
```

Any idea what might be the issue?

Thanks in advance!

jian-dong, 2024-10-24:
Hi @bd-fauna 
The Femto Bolt depth engine seems to require access to GPU resources, which might not be configured correctly in your Docker environment. When you run the command outside of Docker, your system likely has access to the GPU, but this access may not be enabled inside the Docker container.

To fix this issue, please ensure your Docker container is set up to allow access to the GPU. When starting the container, you can check Docker's settings and documentation regarding GPU access, mainly using the `--gpus` option. If GPU access is not correctly configured, the depth engine will not function as expected, leading to errors like yours.

CC @zhonghong322 @xcy2011sky

NickLaurenson-Visionick, 2025-01-22:
Hi,
I encounter the same issue, but solved it by installing the usb rules as mention [here](https://github.com/orbbec/OrbbecSDK/#environment-setup)