orbbec/OrbbecSDK_ROS2 #72: Sync_mode Not Working When Connecting Multiple Femto Bolt Devices

Rhymakers, 2024-11-25:
I am synchronizing four Bolt cameras using the multi_camera_synced.launch.py file.
However, I have identified some issues: modifying the sync_mode parameter in the above file does not apply the changes to the cameras.

As a workaround, I found that setting the sync_mode (Primary, Secondary) through the Orbbec Viewer works correctly.

1. Do Femto Bolt cameras retain the configured parameters (e.g., sync_mode, depth_delay_us) even after they are powered off and on again?

2. Is it correct that the sync_mode setting does not take effect when using the ROS wrapper?

3. Would there be any issues if I set the parameters (sync_mode, depth_delay_us) using the Orbbec Viewer and then proceed with operation? (Assuming only these parameters are adjusted).

I am sharing the code I am currently using. Additionally, regarding the sync_mode parameter, should the input string start with a lowercase letter or an uppercase letter?

Please find the code attached below:
```python
import os
from ament_index_python.packages import get_package_share_directory
from launch import LaunchDescription
from launch.actions import IncludeLaunchDescription, GroupAction, TimerAction
from launch.launch_description_sources import PythonLaunchDescriptionSource

def generate_launch_description():
    # Include launch files
    package_dir = get_package_share_directory('orbbec_camera')
    launch_file_dir = os.path.join(package_dir, 'launch')
    config_file_dir = os.path.join(package_dir, 'config')
    config_file_path = os.path.join(config_file_dir, 'camera_params.yaml')
    front_camera = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(
            os.path.join(launch_file_dir, 'femto_bolt.launch.py')
        ),
        launch_arguments={ 
            'camera_name': 'front_camera',
            'usb_port': '4-3.1',
            'device_num': '4',
            'sync_mode': 'Primary',
            'depth_delay_us' : '0',
            #'config_file_path': config_file_path,
        }.items()
    )

    left_camera = IncludeLaunchDescription( 
        PythonLaunchDescriptionSource(
            os.path.join(launch_file_dir, 'femto_bolt.launch.py')
        ),
        launch_arguments={ 
            'camera_name': 'left_camera',
            'usb_port': '4-3.2',
            'device_num': '4',
            'sync_mode': 'Secondary',
            'depth_delay_us' : '160',
            #'config_file_path': config_file_path,
        }.items()
    )
    rear_camera = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(
            os.path.join(launch_file_dir, 'femto_bolt.launch.py')
        ),
        launch_arguments={ 
            'camera_name': 'rear_camera',
            'usb_port': '4-3.3',
            'device_num': '4',
            'sync_mode': 'Secondary',
            'depth_delay_us' : '320',
            #'config_file_path': config_file_path,
        }.items()
    )
    right_camera = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(
            os.path.join(launch_file_dir, 'femto_bolt.launch.py')
        ),
        launch_arguments={ 
            'camera_name': 'right_camera',
            'usb_port': '4-3.4',
            'device_num': '4',
            'sync_mode': 'Secondary',
            'depth_delay_us' : '480',
            #'config_file_path': config_file_path,
        }.items()
    )

    # If you need more cameras, just add more launch_include here, and change the usb_port and device_num

    # Launch description
    ld = LaunchDescription([
        GroupAction([rear_camera]),
        GroupAction([left_camera]),
        GroupAction([right_camera]),
        TimerAction(period=4.0, actions=[GroupAction([front_camera])]), # The primary camera should be launched at last

    ])

    return ld
```

jjiszjj, 2024-11-27:
@Rhymakers Thank you for using the Femto Bolt camera. You can refer to my answers to the three questions you raised:
1. After the Femto Bolt camera is turned off and then turned on again, it will restore the default configuration and will not retain the last configuration
2. When using ROS wrapper to set the sync_mode mode, the options are standalone, primary, secondary_synced, software_triggering, and hardware_triggering
For multi-camera synchronization, you can visit this website: https://www.orbbec.com/docs/set-up-cameras-for-external-synchronization_v1-2/
I hope it helps you

Rhymakers, 2024-11-27:
@jjiszjj Thank you for anyser. 
2.  Is `secondary` option also available?

jjiszjj, 2024-11-27:
@Rhymakers Thank you for using the Femto Bolt camera. I just asked about the firmware. You can refer to my answers to the three questions you raised:

1. The Femto Bolt camera will save the last camera configuration after it is turned off and then turned on again
2. When using ROS wrapper to set the sync_mode mode, the options are standalone, primary, secondary_synced, software_triggering, and hardware_triggering, you can open the Orbbec Viewer to view the camera mode
For multi-camera synchronization, you can visit this website: https://www.orbbec.com/docs/set-up-cameras-for-external-synchronization_v1-2/
I hope it helps you

Rhymakers, 2024-11-27:
@jjiszjj Thank you for anyser.
2. According to the documentation, the femto model supports only **primary** and **secondary** modes. Can the **secondary** mode also be used in sync mode?
The response doesn't explicitly mention "secondary," so I wanted to ask for clarification

[image]

jjiszjj, 2024-11-28:
@Rhymakers 
I found that the ROS wrapper sets all the sync_mode modes as follows: standalone, primary, secondary_synced, software_triggering, hardware_triggering, free_run, secondary
But I found that the Femto series needs to delete this judgment to successfully set the sync_mode mode. We will fix it in the next version. If you need it urgently, you can modify it yourself
[image]

Rhymakers, 2024-11-29:
@jjiszjj 
Thank you for your solution. The primary and secondary functions are working well in ROS. It would be great if a "syncmode" section could be added to the README.md in the next version update.

closed 2024-11-29 by Rhymakers