orbbec/OrbbecSDK_ROS2 #74: How can I capture the desired moment with a multi-femto bolt camera?

Rhymakers, 2024-11-29:
I have set up an environment where four femto bolt cameras are connected to a Sync Hun Pro. What I want is to capture RGB and depth (without IR interference) at a specific moment from all four cameras. I have two methods in mind, and I'd like to perform one of them, but I'm facing some difficulties. Hence, I’m posting this question.

Configure 1 primary and 3 secondary cameras, and control the on/off of RGB and depth on the primary camera.
Configure 4 secondary cameras and use a PC as the primary to receive RGB and depth.
When attempting method 1, I found that the ROS source doesn’t have an option to capture a single frame from a specific camera. For method 2, it seems that I need to write an Ethernet protocol using ROS.

Which method would be better, and is there a solution?

3Ddaiwei, 2024-12-05:
It is recommended to use configuration 1. The time synchronization through master-slave synchronization will be relatively more accurate. For each stream of the camera, the ros-wrapper will publish corresponding topics. You can listen to the topic of the data stream frame to be captured and save the frame data.

Rhymakers, 2024-12-05:
@3Ddaiwei Thank you for your response. You recommended Configuration 1, but I plan to keep the system running for a long time and would like the cameras to send data only when needed. Of course, I could always leave the cameras on and use the data only when required. However, I would prefer to turn on the primary camera when I need it, which in turn activates the data transmission from the three secondary cameras, and then turn off the primary camera afterward.

Is there a ROS service or action already implemented for this purpose? If not, could you share some ideas on how I could implement this?