orbbec/OrbbecSDK_ROS2 #79: Error on camera info topics retrieval in ROS2

Anil-Bhujel, 2024-12-12:
Hi,
We have 6 Femto Mega camera systems to collect the video data. We want to store the color and depth topic along with color and depth camera info parameter in ROS2 bag. We have used network Femto mega in our system. We can see the ROS2 topic names for color, depth, and IR camera info parameters, but when we try to echo or record the camera parameters, the error raised and the camera node has been died. Please suggest how can we get it for recording.

[image]

jian-dong, 2024-12-13:
Can you please paste the crash log? It's located in `Log/camera_01_crash_stack_xxx`.

Anil-Bhujel, 2024-12-13:
@jian-dong ,
Thank you for your prompt response. Please find the attached log files.
[camera_01_crash_stack_trace_2024_12_11_18_32_25.log](https://github.com/user-attachments/files/18131159/camera_01_crash_stack_trace_2024_12_11_18_32_25.log)
[camera_01_crash_stack_trace_2024_12_11_18_32_26.log](https://github.com/user-attachments/files/18131162/camera_01_crash_stack_trace_2024_12_11_18_32_26.log)

Anil-Bhujel, 2024-12-16:
@jian-dong 
Received signal: 11
Stack trace (most recent call last) in thread 101532:
#8    Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
#7    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7d4819daba3f, in 
#6    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7d4819d19ac2, in 
#5    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7d4819fa9252, in 
#4    Object "/workspaces/isaac_ros-dev/install/orbbec_camera/lib/liborbbec_camera.so", at 0x7d48143dbf41, in orbbec_camera::OBCameraNode::onNewColorFrameCallback()
#3    Object "/workspaces/isaac_ros-dev/install/orbbec_camera/lib/liborbbec_camera.so", at 0x7d48143d42f8, in orbbec_camera::OBCameraNode::onNewFrameCallback(std::shared_ptr<ob::Frame> const&, std::pair<OBStreamType, int> const&)
#2    Object "/opt/ros/humble/lib/librclcpp.so", at 0x7d481a281f1f, in rclcpp::PublisherBase::get_subscription_count() const
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7d4819cc751f, in 
#0    Object "/workspaces/isaac_ros-dev/install/orbbec_camera/lib/liborbbec_camera.so", at 0x7d4814388e9d, in signalHandler(int)

jjiszjj, 2024-12-20:
@Anil-Bhujel 
Regarding the issue of Error on camera info topics retrieval in ROS2, I have fixed it. You can refer to these two commit hashes to see the changes:
35405405aaa6320bb5650b10cbfa4912c9ae1153
997ff451ade359d12884cb0293bc468966b9cd3a

Anil-Bhujel, 2024-12-21:
@jian-dong and @jjiszjj 
Thank you so much. It worked

closed 2024-12-21 by Anil-Bhujel