orbbec/OrbbecSDK_ROS2 #80: v2-main branch failed to initialize device.

graycatrl, 2024-12-13:
Issue: Gemini 336 camera works with `main` branch but not `v2-main` branch.

Setup:
- Ubuntu 22.04
- ROS2 Humble fresh install 
- Cloned and built this repo following instructions in README.

If I'm in main branch, camera is able to initialize:
```
$ ros2 launch orbbec_camera gemini_330_series.launch.py
[INFO] [launch]: All log files can be found below /home/rl/.ros/log/2024-12-12-20-59-15-952849-rl-ubuntu-68366
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [68378]
[component_container-1] [INFO] [1734065956.260821483] [camera.camera_container]: Load Library: /home/rl/ros2_ws/install/orbbec_camera/lib/liborbbec_camera.so
[component_container-1] [INFO] [1734065956.296795103] [camera.camera_container]: Found class: rclcpp_components::NodeFactoryTemplate<orbbec_camera::OBCameraNodeDriver>
[component_container-1] [INFO] [1734065956.296823045] [camera.camera_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<orbbec_camera::OBCameraNodeDriver>
[component_container-1] [12/12 20:59:16.299357][info][68378][Context.cpp:68] Context created with config: /home/rl/ros2_ws/install/orbbec_camera/share/orbbec_camera/config/OrbbecSDKConfig_v1.0.xml
[component_container-1] [12/12 20:59:16.299368][info][68378][Context.cpp:73] Work directory=/home/rl/ros2_ws, SDK version=v1.10.16-20241021-5113dad
[component_container-1] [12/12 20:59:16.299378][info][68378][LinuxPal.cpp:32] createObPal: create LinuxPal!
[component_container-1] [12/12 20:59:16.376901][info][68378][LinuxPal.cpp:166] Create PollingDeviceWatcher!
[component_container-1] [12/12 20:59:16.376925][info][68378][DeviceManager.cpp:15] Current found device(s): (1)
[component_container-1] [12/12 20:59:16.376929][info][68378][DeviceManager.cpp:24] 	- Name: Orbbec Gemini 336, PID: 0x0803, SN/ID: CP7284100039, Connection: USB3.2
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/camera/camera' in container '/camera/camera_container'
[component_container-1] [INFO] [1734065956.387861892] [camera.camera]: Connecting to the default device
[component_container-1] [INFO] [1734065956.496731385] [camera.camera]: Select device cost 108 ms
[component_container-1] [INFO] [1734065956.496802287] [camera.camera]: Try to connect device via USB3.2
[component_container-1] [INFO] [1734065956.496875643] [camera.camera]: OBCameraNode: use_intra_process: OFF
[component_container-1] [INFO] [1734065956.498249724] [camera.camera]: current time domain: device
[component_container-1] [INFO] [1734065956.502504453] [camera.camera]: Setting heartbeat to OFF
[component_container-1] [INFO] [1734065956.502668137] [camera.camera]: Create align filter
[component_container-1] [INFO] [1734065956.502942416] [camera.camera]: Depth process is HW D2D
[component_container-1] [INFO] [1734065956.502947606] [camera.camera]: Setting LDP to ON
[component_container-1] [INFO] [1734065956.503071496] [camera.camera]: Setting laser control to 1
[component_container-1] [INFO] [1734065956.504333179] [camera.camera]: Setting laser on off mode to 0
[component_container-1] [INFO] [1734065956.504455456] [camera.camera]: Available presets:
[component_container-1] [INFO] [1734065956.504466416] [camera.camera]: Preset 0: Custom
[component_container-1] [INFO] [1734065956.504470073] [camera.camera]: Preset 1: Default
[component_container-1] [INFO] [1734065956.504473659] [camera.camera]: Preset 2: AMR with IR-Pass
[component_container-1] [INFO] [1734065956.504476735] [camera.camera]: Preset 3: Factory Calib
[component_container-1] [INFO] [1734065956.504480061] [camera.camera]: Load device preset: Default
[component_container-1] [INFO] [1734065956.504491583] [camera.camera]: Device preset Default loaded
[component_container-1] [INFO] [1734065956.504651850] [camera.camera]: Current sync mode: OB_MULTI_DEVICE_SYNC_MODE_STANDALONE
[component_container-1] [INFO] [1734065956.504659945] [camera.camera]: Set sync mode: OB_MULTI_DEVICE_SYNC_MODE_STANDALONE
[component_container-1] [INFO] [1734065956.504664143] [camera.camera]: Setting color auto white balance to ON
[component_container-1] [INFO] [1734065956.505968915] [camera.camera]: Setting color auto exposure to ON
[component_container-1] [INFO] [1734065956.507306048] [camera.camera]: Setting IR auto exposure to ON
[component_container-1] [INFO] [1734065956.512459165] [camera.camera]: Setting DecimationFilter......
[component_container-1] [INFO] [1734065956.512471148] [camera.camera]: set DecimationFilter to false
[component_container-1] [INFO] [1734065956.512482929] [camera.camera]: Skip setting filter: DecimationFilter
[component_container-1] [INFO] [1734065956.512488259] [camera.camera]: Setting HDRMerge......
[component_container-1] [INFO] [1734065956.512491806] [camera.camera]: set HDRMerge to false
[component_container-1] [INFO] [1734065956.512495302] [camera.camera]: Skip setting filter: HDRMerge
[component_container-1] [INFO] [1734065956.512499190] [camera.camera]: Setting SequenceIdFilter......
[component_container-1] [INFO] [1734065956.512502416] [camera.camera]: Skip setting filter: SequenceIdFilter
[component_container-1] [INFO] [1734065956.512506022] [camera.camera]: Setting NoiseRemovalFilter......
[component_container-1] [INFO] [1734065956.512509058] [camera.camera]: set NoiseRemovalFilter to true
[component_container-1] [INFO] [1734065956.512517904] [camera.camera]: Default noise removal filter params: disp_diff: 256, max_size: 80
[component_container-1] [INFO] [1734065956.512521321] [camera.camera]: Set noise removal filter params: disp_diff: 256, max_size: 80
[component_container-1] [INFO] [1734065956.512526210] [camera.camera]: Setting SpatialAdvancedFilter......
[component_container-1] [INFO] [1734065956.512529706] [camera.camera]: set SpatialAdvancedFilter to false
[component_container-1] [INFO] [1734065956.512532872] [camera.camera]: Skip setting filter: SpatialAdvancedFilter
[component_container-1] [INFO] [1734065956.512536409] [camera.camera]: Setting SpatialFastFilter......
[component_container-1] [INFO] [1734065956.512539434] [camera.camera]: Skip setting filter: SpatialFastFilter
[component_container-1] [INFO] [1734065956.512542821] [camera.camera]: Setting SpatialModerateFilter......
[component_container-1] [INFO] [1734065956.512545676] [camera.camera]: Skip setting filter: SpatialModerateFilter
[component_container-1] [INFO] [1734065956.512549022] [camera.camera]: Setting TemporalFilter......
[component_container-1] [INFO] [1734065956.512552158] [camera.camera]: set TemporalFilter to false
[component_container-1] [INFO] [1734065956.512555454] [camera.camera]: Skip setting filter: TemporalFilter
[component_container-1] [INFO] [1734065956.512559271] [camera.camera]: Setting HoleFillingFilter......
[component_container-1] [INFO] [1734065956.512562838] [camera.camera]: set HoleFillingFilter to false
[... 228 lines cut ...]
[component_container-1] [INFO] [1734065564.738073131] [camera.camera]: Set sync mode: OB_MULTI_DEVICE_SYNC_MODE_STANDALONE
[component_container-1] [ERROR] [1734065564.738095102] [camera.camera]: Failed to setup topics: Filter@FrameProcessor: config schema is empty, doesn't have any config value
[component_container-1] [ERROR] [1734065564.738121301] [camera.camera]: Failed to initialize device (Attempt 2 of 3): Filter@FrameProcessor: config schema is empty, doesn't have any config value
[component_container-1] [INFO] [1734065564.738144564] [camera.camera]: OBCameraNode: use_intra_process: OFF
[component_container-1] [INFO] [1734065564.738206300] [camera.camera]: current time domain: device
[component_container-1] [INFO] [1734065564.739289033] [camera.camera]: Setting heartbeat to OFF
[component_container-1] [INFO] [1734065564.739304402] [camera.camera]: Create align filter
[component_container-1] [INFO] [1734065564.739470112] [camera.camera]: Depth process is HW D2D
[component_container-1] [INFO] [1734065564.739475051] [camera.camera]: Setting LDP to ON
[component_container-1] [INFO] [1734065564.739606907] [camera.camera]: Setting G300 laser control to 1
[component_container-1] [INFO] [1734065564.739740787] [camera.camera]: Setting laser on off mode to 0
[component_container-1] [INFO] [1734065564.739867735] [camera.camera]: Available presets:
[component_container-1] [INFO] [1734065564.739872153] [camera.camera]: Preset 0: Default
[component_container-1] [INFO] [1734065564.739875068] [camera.camera]: Preset 1: AMR with IR-Pass
[... cut ...]
[component_container-1] [ERROR] [1734065564.741302697] [camera.camera]: Failed to setup topics: Filter@FrameProcessor: config schema is empty, doesn't have any config value
[component_container-1] [ERROR] [1734065564.741325680] [camera.camera]: Failed to initialize device (Attempt 3 of 3): Filter@FrameProcessor: config schema is empty, doesn't have any config value
[component_container-1] [ERROR] [1734065564.741329226] [camera.camera]: Device initialization failed after 3 attempts.
[component_container-1] [ERROR] [1734065564.741340678] [camera.camera]: Failed to initialize device Device initialization failed after 3 attempts.
```

jjiszjj, 2024-12-19:
@graycatrl Thank you for your feedback. If you can delete the build and install folders and then switch to v2-main and recompile, will there still be a problem of being unable to initialize the device?

graycatrl, 2024-12-21:
> @graycatrl Thank you for your feedback. If you can delete the build and install folders and then switch to v2-main and recompile, will there still be a problem of being unable to initialize the device?

@jjiszjj It seems the previous workaround is no longer working for me. I have tried deleting build/install folders between rebuilds when switching between main/v2-main branch, and v2-main branch produce the same error while it's running fine in main branch. Even if I don't delete the build/install folders between rebuilds, v2-main still won't work.

FabienDanieau, 2025-02-06:
Hey, I have the exact same problem. Lastest v2-main does not fix the issue.

bachev93, 2025-02-10:
Hello, I have the same problem. I build and run femto bolt camera on "Jazzy" successfully, but on "Humble" femto bolt camera has runtime error "Failed to initialize device".

bachev93, 2025-03-27:
I found workaround. I build ROS2 humble from sources in Dockerfile, based on Ubuntu 24.04. Looks like, that Orbbec SDK ver.2 doesn't correctly work on Ubuntu 22.04. My Dockerfile:

`
FROM ubuntu:noble

# setup timezone
RUN echo 'Etc/UTC' > /etc/timezone && \
    ln -s /usr/share/zoneinfo/Etc/UTC /etc/localtime && \
    apt-get update && \
    apt-get install -q -y --no-install-recommends tzdata && \
    rm -rf /var/lib/apt/lists/*

RUN apt update && apt install -y locales && \
    locale-gen en_US en_US.UTF-8 && \
    update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8 && \
    rm -rf /var/lib/apt/lists/*

ENV LANG=en_US.UTF-8

RUN apt update && apt install -y software-properties-common && \
    add-apt-repository universe && \
    rm -rf /var/lib/apt/lists/*

RUN apt update && apt install curl -y && \
    curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key -o /usr/share/keyrings/ros-archive-keyring.gpg

RUN echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu $(. /etc/os-release && echo jammy) main" | tee /etc/apt/sources.list.d/ros2.list > /dev/null

RUN apt update && apt install -y \
    python3-flake8-docstrings \
    python3-pip \
    python3-pytest-cov \
    ros-dev-tools \
    python3-flake8-blind-except \
    python3-flake8-builtins \
    python3-flake8-class-newline \
    python3-flake8-comprehensions \
    python3-flake8-deprecated \
    python3-flake8-import-order \
    python3-flake8-quotes \
    python3-pytest-repeat \
    python3-pytest-rerunfailures && \
    rm -rf /var/lib/apt/lists/*

RUN apt update && apt install -y \
    libboost1.83-all-dev \
    libogg-dev \
    libtheora-dev \
    libgflags-dev \
    nlohmann-json3-dev \
    libdw-dev && \
    rm -rf /var/lib/apt/lists/*

# ROS2 Humble installation
ENV ROS_DISTRO=humble
WORKDIR /ros2_humble
RUN mkdir src && vcs import --input https://raw.githubusercontent.com/ros2/ros2/humble/ros2.repos src
RUN rm -rf ./src/ignition/ ./src/ros2/rviz/ ./src/ros-visualization
RUN apt update && rosdep init && rosdep update && \
    rosdep install --from-paths src --ignore-src -y --skip-keys "fastcdr rti-connext-dds-6.0.1 urdfdom_headers" && \
    rm -rf /var/lib/apt/lists/*
RUN cd src/ros-perception && git clone https://github.com/ros-perception/image_transport_plugins.git -b humble \
    && git clone https://github.com/ros-perception/image_pipeline.git -b humble \
    && git clone https://github.com/ros-perception/vision_opencv.git -b humble
RUN cd src && git clone https://github.com/ros/diagnostics.git -b ros2-humble \
    && git clone https://github.com/pal-robotics/backward_ros.git -b foxy-devel
RUN colcon build --symlink-install

ARG USER=docker_orbbec
ARG UID=1000
ARG GID=1000
ARG PW=user

RUN userdel -r ubuntu
RUN useradd -m ${USER} --uid=${UID} && echo "${USER}:${PW}" | chpasswd && adduser ${USER} sudo
WORKDIR /home/${USER}
RUN mkdir -p ros2_ws/src && chown -R ${UID}:${GID} /home/${USER}

WORKDIR /home/${USER}/ros2_ws/src
COPY . ./OrbbecSDK_ROS2

# Build
RUN cd .. && . /ros2_humble/install/setup.sh && \
    colcon build --event-handlers  console_direct+ --cmake-args -DCMAKE_BUILD_TYPE=Release

WORKDIR /home/${USER}
`

mzahana, 2025-04-03:
I am facing the same problem with the 335gl camera

I used the command
```bahs
ros2 launch orbbec_camera gemini_330_series.launch.py

```

and I got,

```bash
[component_container-1] 
[component_container-1] [ERROR] [1743690704.619130387] [camera.camera]: Failed to setup topics: HidDevicePortGmsl() openDev failed, errno: 2 No such file or directory
[component_container-1] [ERROR] [1743690704.619502614] [camera.camera]: Failed to initialize device (Attempt 1 of 3): HidDevicePortGmsl() openDev failed, errno: 2 No such file or directory
[component_container-1] [INFO] [1743690704.619755800] [camera.camera]: OBCameraNode: use_intra_process: OFF
[component_container-1] [INFO] [1743690704.620604990] [camera.camera]: current time domain: global
[component_container-1] [INFO] [1743690704.620703103] [camera.camera]: hdr_index1_laser_control_ 1 hdr_index1_depth_exposure_ 1 hdr_index1_depth_gain_ 16 hdr_index1_ir_brightness_ 20 hdr_index1_ir_ae_max_exposure_ 2000
[component_container-1] 
[component_container-1] [INFO] [1743690704.620753727] [camera.camera]: hdr_index0_laser_control_ 1 hdr_index0_depth_exposure_ 7500 hdr_index0_depth_gain_ 16 hdr_index0_ir_brightness_ 60 hdr_index0_ir_ae_max_exposure_ 10000
[component_container-1] 
[component_container-1] [INFO] [1743690704.620791167] [camera.camera]: laser_index1_laser_control_ 0 laser_index1_depth_exposure_ 3000 laser_index1_depth_gain_ 16 laser_index1_ir_brightness_ 60 laser_index1_ir_ae_max_exposure_ 17000
[component_container-1] 
[component_container-1] [INFO] [1743690704.620823744] [camera.camera]: laser_index0_laser_control_ 1 laser_index0_depth_exposure_ 3000 laser_index0_depth_gain_ 16 laser_index0_ir_brightness_ 60 laser_index0_ir_ae_max_exposure_ 30000
[component_container-1] 
[component_container-1] [ERROR] [1743690704.628965916] [camera.camera]: Failed to setup topics: Component 14 not found!
[component_container-1] [ERROR] [1743690704.629275518] [camera.camera]: Failed to initialize device (Attempt 2 of 3): Component 14 not found!
[component_container-1] [INFO] [1743690704.629504960] [camera.camera]: OBCameraNode: use_intra_process: OFF
[component_container-1] [INFO] [1743690704.630094821] [camera.camera]: current time domain: global
[component_container-1] [INFO] [1743690704.630171781] [camera.camera]: hdr_index1_laser_control_ 1 hdr_index1_depth_exposure_ 1 hdr_index1_depth_gain_ 16 hdr_index1_ir_brightness_ 20 hdr_index1_ir_ae_max_exposure_ 2000
[component_container-1] 
[component_container-1] [INFO] [1743690704.630220454] [camera.camera]: hdr_index0_laser_control_ 1 hdr_index0_depth_exposure_ 7500 hdr_index0_depth_gain_ 16 hdr_index0_ir_brightness_ 60 hdr_index0_ir_ae_max_exposure_ 10000
[component_container-1] 
[component_container-1] [INFO] [1743690704.630262566] [camera.camera]: laser_index1_laser_control_ 0 laser_index1_depth_exposure_ 3000 laser_index1_depth_gain_ 16 laser_index1_ir_brightness_ 60 laser_index1_ir_ae_max_exposure_ 17000
[component_container-1] 
[component_container-1] [INFO] [1743690704.630297542] [camera.camera]: laser_index0_laser_control_ 1 laser_index0_depth_exposure_ 3000 laser_index0_depth_gain_ 16 laser_index0_ir_brightness_ 60 laser_index0_ir_ae_max_exposure_ 30000
[component_container-1] 
[component_container-1] [ERROR] [1743690704.637238522] [camera.camera]: Failed to setup topics: Component 14 not found!
[component_container-1] [ERROR] [1743690704.637409563] [camera.camera]: Failed to initialize device (Attempt 3 of 3): Component 14 not found!
[component_container-1] [ERROR] [1743690704.637464699] [camera.camera]: Device initialization failed after 3 attempts.
[component_container-1] [ERROR] [1743690704.637520764] [camera.camera]: Failed to initialize device Device initialization failed after 3 attempts.
[component_container-1] [INFO] [1743690704.637621661] [camera.camera]: resetDevice : Reset device uid: 
[component_container-1] [INFO] [1743690704.637800926] [camera.camera]: Reset device uid:  done
[component_container-1] [INFO] [1743690704.648733071] [camera.camera]: Connecting to the default device
[component_container-1] [INFO] [1743690705.213222775] [camera.camera]: Select device cost 564 ms
[component_container-1] [INFO] [1743690705.213448665] [camera.camera]: Try to connect device via GMSL2
[component_container-1] [INFO] [1743690705.213646842] [camera.camera]: OBCameraNode: use_intra_process: OFF
[component_container-1] [INFO] [1743690705.214420736] [camera.camera]: current time domain: global
[component_container-1] [INFO] [1743690705.214510656] [camera.camera]: hdr_index1_laser_control_ 1 hdr_index1_depth_exposure_ 1 hdr_index1_depth_gain_ 16 hdr_index1_ir_brightness_ 20 hdr_index1_ir_ae_max_exposure_ 2000
[component_container-1] 
[component_container-1] [INFO] [1743690705.214570241] [camera.camera]: hdr_index0_laser_control_ 1 hdr_index0_depth_exposure_ 7500 hdr_index0_depth_gain_ 16 hdr_index0_ir_brightness_ 60 hdr_index0_ir_ae_max_exposure_ 10000
[component_container-1] 
[component_container-1] [INFO] [1743690705.214615425] [camera.camera]: laser_index1_laser_control_ 0 laser_index1_depth_exposure_ 3000 laser_index1_depth_gain_ 16 laser_index1_ir_brightness_ 60 laser_index1_ir_ae_max_exposure_ 17000
[component_container-1] 
[component_container-1] [INFO] [1743690705.214648290] [camera.camera]: laser_index0_laser_control_ 1 laser_index0_depth_exposure_ 3000 laser_index0_depth_gain_ 16 laser_index0_ir_brightness_ 60 laser_index0_ir_ae_max_exposure_ 30000
[component_container-1] 
[... cut ...]
[component_container-1] [ERROR] [1743690705.342462808] [camera.camera]: Failed to initialize device Device initialization failed after 3 attempts.
[component_container-1] [INFO] [1743690705.342570713] [camera.camera]: resetDevice : Reset device uid: 
[component_container-1] [INFO] [1743690705.342740091] [camera.camera]: Reset device uid:  done
[component_container-1] [INFO] [1743690705.353863341] [camera.camera]: Connecting to the default device

```

I am using the `v2-main` branch. I also deleted the build/install directories, but the issue persists.

I have to use ROS 2 `humble`.

@jjiszjj Any solutions ?

mzahana, 2025-04-04:
My issue was solved after adding `--privileged` to the docker run command.