orbbec/OrbbecSDK_ROS2 #83: v2-main branch does not initialize filters.

john-maidbot, 2024-12-17:
Hello,

At first I noticed that decimation filter did not seem to be working. When I looked closer, no filters work atm on v2-main. 
It appears there is a typo (https://github.com/orbbec/OrbbecSDK_ROS2/blob/0b0474996173efe64126a9cede44d193c2b4b9c3/orbbec_camera/src/ob_camera_node.cpp#L474) that prevents the filters from being setup. It is returning early if the list is not empty. but it should be the reverse.

OS:
Ubuntu 22.04

Model:
Gemini 335

Software:
OrbbecSDK_ROS2, v2-main branch

Node parameters checked at runtime:
    decimation_filter_scale_: 2
    enable_decimation_filter: true
    depth_camera_info_qos: default
    depth_format: ANY
    depth_fps: 10
    depth_height: 480
    depth_registration: false
    depth_width: 640
    enable_depth: true
    enable_depth_scale: true

Expectation:

The published depth image should be 320 x 240 (i.e. 640 / 2 x 480 / 2)

Actual behavior:
```
ros2 topic echo /camera/depth/image_raw
```

```
header:
  stamp:
    sec: 1734464379
    nanosec: 723162880
  frame_id: camera_depth_optical_frame
height: 480
width: 640
encoding: 16UC1
is_bigendian: 0
step: 1280
data:
...
```

I can make a PR if that would be helpful?

Thank you for your time.

jjiszjj, 2024-12-19:
@john-maidbot Thanks for the feedback, I will do a check on v2-main

jjiszjj, 2024-12-19:
@john-maidbot After our debugging, we found that the problem you described does exist. The modification you described is correct. Thank you for your feedback during use. We will fix this error in the next version update.

closed 2025-11-25 by ob-yalian