orbbec/OrbbecSDK_ROS2 #86: Difference in Camera Depth Intrinsic Parameters Between Older and Newer Versions

Rhymakers, 2024-12-27:
I was using the ROS2 wrapper supporting SDK1 version, but recently switched to the ROS2 wrapper supporting SDK2 version. As a result, I noticed that the depth intrinsic parameters in the older and newer version wrappers are different, as shown below. I have enabled the registration option for 3D point cloud generation. Which one is the correct state?

[image]
Above is the older version, Below is the newer version
[image]

jian-dong, 2024-12-27:
Hi @Rhymakers 
Can you please tell me which camera you are using?

Rhymakers, 2024-12-27:
Hi @jian-dong  femto bolt!