orbbec/OrbbecSDK_ROS2 #87: cannot launch

Tang-Yihang, 2025-01-05:
I have error as follow
[component_container-1] Received signal: 11
[component_container-1] Log crash stack trace to Log/camera_crash_stack_trace_2025_01_05_15_09_02.log
[component_container-1] Received signal: 11
[component_container-1] Log crash stack trace to Log/camera_crash_stack_trace_2025_01_05_15_09_02.log
[ERROR] [component_container-1]: process has died [pid 397357, exit code 11, cmd '/opt/ros/humble/lib/rclcpp_components/component_container --ros-args -r __node:=camera_container -r __ns:=/camera'].

ricardohogan, 2025-01-08:
I also faced with similar problem with my newly bought femton bolt

jjiszjj, 2025-01-08:
@Tang-Yihang @ricardohogan Thank you for your feedback. I tried to reproduce your phenomenon in the lab, but failed. If you can provide

jjiszjj, 2025-01-08:
@Tang-Yihang If you can provide camera_crash_stack_trace_2025_01_05_15_09_02.log, it will be very helpful for us to analyze the problem.

ricardohogan, 2025-01-09:
I've managed to solve this issue by changing the USB I realized that different USB port on my computer. 
The error sometimes still exist though, but the frequency is a lot less. 
I've also tested on 2 other computers running Ubuntu 20.04 (the same with first computer that I tested): 1 jetson orin & 1 another desktop.
On jetson orin, the error appears rarely, and on the other desktop the error never appears.

So I guess there's an incompatibility between my first USB port and the camera, but for now I assumed this issue as solved.

closed 2025-11-25 by obshenlanzhou