orbbec/OrbbecSDK_ROS2 #88: Feature Request: imu offset compensation service

john-maidbot, 2025-01-11:
I find that the imu in my orbbec sensor is accurate but often has an initial offset in angular velocity and linear acceleration that needs to be corrected. I see this behavior on both femto and gemini-335 (but they have different offsets)

i see that you already correct for bias accumulation in the ros2 wrapper, but it would be nice to also have an option to correct for this initial offset (eg place sensor on level surface and call a ros2 service. Thats what i am doing on my end atm, but it would be nice if this was in the orbbec node. Plus i figure others would benefit from this too). Have you considered adding such a feature? 

thank you for your time

jjiszjj, 2025-02-19:
@john-maidbot Hello,thank you very much for your suggestion.From what I understand from the text you said, do you mean to add a new service function to recalibrate the initial position of imu?

john-maidbot, 2025-02-19:
Thank you for your reply. 

The intention of the service would be to recalibrate the acceleration and angular velocity bias offsets of the imu while level and at rest. Not the position.

Ie. While at rest and level, imu angular velocity should be nearly (0,0,0) rad/s and acceleration should be nearly (0,0,-9.81) m/s^2 (assuming imu z axis is aligned with gravity for simplicity here). And currently that is often not the case when i start the cameras