orbbec/OrbbecSDK_ROS2 #92: ROS2 orbbec_description launch not working for femto bolt.

MaartenG18, 2025-02-16:
The launch command in the readme file underneath https://github.com/orbbec/OrbbecSDK_ROS2/tree/v2-main?tab=readme-ov-file#tf-from-coordinate-a-to-coordinate-b is not working if you use the femto_bolt urdf model. 

ros2 launch orbbec_description view_model.launch.py model:=femto_bolt.urdf.xacro
[INFO] [launch]: All log files can be found below /home/maarten/.ros/log/2025-02-16-14-45-00-337769-maarten-HP-EliteBook-840-G5-48502
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [48517]
[INFO] [robot_state_publisher-2]: process started with pid [48518]
[robot_state_publisher-2] [WARN] [1739713500.558088859] [model_node]: No robot_description parameter, but command-line argument available.  Assuming argument is name of URDF file.  This backwards compatibility fallback will be removed in the future.
[robot_state_publisher-2] Error:   No link elements found in urdf file
[robot_state_publisher-2]          at line 207 in ./urdf_parser/src/model.cpp
[robot_state_publisher-2] Failed to parse robot description using: urdf_xml_parser/URDFXMLParser
[robot_state_publisher-2] terminate called after throwing an instance of 'std::runtime_error'
[robot_state_publisher-2]   what():  Unable to initialize urdf::model from robot description
[rviz2-1] [INFO] [1739713501.126496999] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [INFO] [1739713501.126936080] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-1] [INFO] [1739713501.188603239] [rviz2]: Stereo is NOT SUPPORTED
[ERROR] [robot_state_publisher-2]: process has died [pid 48518, exit code -6, cmd '/opt/ros/jazzy/lib/robot_state_publisher/robot_state_publisher /tmp/femto_bolt.urdf.xacro_68xlym0g --ros-args -r node:=model_node -r ns:=/'].
^[[A^[[A^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[rviz2-1] [INFO] [1739713541.712386486] [rclcpp]: signal_handler(signum=2)
[INFO] [rviz2-1]: process has finished cleanly [pid 48517]

jjiszjj, 2025-02-19:
@MaartenG18 Hello,Thank you very much for your feedback.There is indeed a problem with Femto_bolt in orbbec_description. We are considering whether to delete it or replace it with a new model.

MaartenG18, 2025-02-19:
@jjiszjj

I understand the consideration to remove the camera_link TF, but I believe it still serves a useful purpose in representing the 3D model of the robot. If you decide to remove it, I strongly recommend providing a clear statement explaining what the camera_link TF represents and its orientation.

From my research (though I am still somewhat uncertain), it seems that camera_link is defined inside the camera itself. In the file femto_bolt.urdf.xacro, I found the coordinates on lines 21–23:

```xml
<xacro:property name="base_link_to_link_x" value="0.03645"/> <xacro:property name="base_link_to_link_y" value="0.00198"/> <xacro:property name="base_link_to_link_z" value="0.021"/> 
```

These values could be used to transform camera_link to the mounting screw. Having a well-defined transformation from the internal reference of the camera to an external mounting point is particularly useful in multi-camera setups, yet this information is rarely documented.

Please keep this in mind when considering the removal of the model, as such details could be valuable for users working with multiple cameras.

jjiszjj, 2025-02-19:
@MaartenG18 Yes.In robot simulation, the camera_link of the 3D model should be defined inside the camera. However, the current orbbec_description only provides the femto_bolt.stl shell model, so we think this is a problem and plan to update it in the next version. In addition, the developer ideas and suggestions you provided are very helpful, and we will refer to them and improve them.

MaartenG18, 2025-02-19:
@jjiszjj Thank you for your help and I am already looking forward to the improvements!

scolianaris, 2025-11-25:
Hi @MaartenG18,

Thanks again for your feedback. We’ve already improved the URDF model for Femto Bolt in tag v2.5.5, including fixing the structure issues and updating the TF definitions.

If you update to v2.5.5, the model should load correctly in orbbec_description.
Please let us know if you encounter any further problems.