orbbec/OrbbecSDK_ROS2 #93: gemini330 ros2 depth cloud point  result_frame->dataSize() not stable

tarikbeijing, 2025-02-25:
it is about this void : 
      void OBCameraNode::publishDepthPointCloud(const std::shared_ptr<ob::FrameSet> &frame_set) {

i see that the result frame when i display it crop regularly. i have follow back the data
i see coming from camera
      auto point_size = result_frame->dataSize() / sizeof(OBPoint);
 
dadasize 12288000
dadasize 12288000
dadasize 12288000
dadasize 11059200<<<<<----
dadasize 12288000
dadasize 12288000
dadasize 11059200<<<<<<---

later when i look the valid count
point_cloud_msg->width = valid_count;

i have also the same effect 

valid_count2 : 909707
valid_count2 : 913847
valid_count2 : 911681
valid_count2 : 737903<<<<<----
valid_count2 : 909707
valid_count2 : 913847
valid_count2 : 913647

when i display the result it look like that the max opening angle regulary is not the same

it disturbing a lot the result. 

in the opposite the color_depth frame dont have that problem

i used 2 different gemini 330, both has this problem, i use the last firmware and  ros2 driver, lunux ubuntu 22.04

jjiszjj, 2025-03-11:
@tarikbeijing Hello,Thank you very much for your feedback.
Can you provide me with the OrbbecSDK_ROS2 version you are using and the firmware information of gemini330?

jjiszjj, 2025-03-11:
@tarikbeijing Hello,Thank you very much for your feedback.
Can you provide me with the specific Gemini 330 device, firmware information and OrbbecSDK_ROS2 version number?

tarikbeijing, 2025-06-24:
so it look  this is a problem from SDK1, i have now switch on Ros_sdk_version : 2.4.3 ( no change done on the firmware)
Device Orbbec Gemini 335 connected
Firmware version: 1.4.60
Hardware version: 0.1
Ros_sdk_version : 2.4.3
ob_sdk_version : 2.4.3
usb connect type: USB3.2

the problem is solved, the valid_count is stable now