orbbec/OrbbecSDK_ROS2 #94: Gemini 2L Empty camera info at 1280x800 resolution

pragun-ocado, 2025-03-04:
Similar to https://github.com/orbbec/OrbbecSDK_ROS2/issues/65

We're seeing missing values for fx, fy, cx, cy, etc in camera_info for the Gemini 2L at 1280x800px.

```
header:
  stamp:
    sec: 1741117573
    nanosec: 274824960
  frame_id: camera_mast_rear_downward_color_optical_frame
height: 800
width: 1280
distortion_model: rational_polynomial
d:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
k:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 1.0
r:
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
p:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 1.0
- 0.0
binning_x: 0
binning_y: 0
roi:
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: false
---
```

pragun-ocado, 2025-03-04:
Furthermore, we notice that if the depth stream is turned on the camera_info is populated correctly for the color stream as well

jjiszjj, 2025-03-11:
@pragun-ocado Hello,Thank you very much for your feedback.
Could you provide me with the specific Gemini 2 L device, firmware information and OrbbecSDK_ROS2 version number? This will help me reproduce and debug the issue in my lab. Or you can try to update [firmware](https://github.com/orbbec/OrbbecFirmware/releases/tag/Gemini2L-Firmware) and [OrbbecSDK_ROS2 v2.2.3](https://github.com/orbbec/OrbbecSDK_ROS2/releases/tag/v2.2.3)