orbbec/OrbbecSDK_ROS2 #97: Can't set FPS on Femto Mega

kazu-321, 2025-03-16:
Hello!!!
Femto Mega is connected via USB or ETH
I was able to set the FPS properly with Orbbec viewer, but it doesn't work when using ROS 2.
Even though I set the color_fps value to 5, it actually gets 30fps.
The environment and logs are as follows

Ubuntu 22.04
ROS 2 humble
SDK,Firmware: latest
commit id: 052846f9ef1260d34c3b9809a4d8bb625bb112f0

[log.log](https://github.com/user-attachments/files/19268215/log.log)

kazu-321, 2025-03-16:
https://github.com/orbbec/OrbbecSDK_ROS2/tree/femto_mega_h26x_decode
I just found this branch so I'll give it a try.

kazu-321, 2025-03-17:
@jjiszjj Thank you for your fast response!
I encountered an error at startup when enable_frame_sync was set to true, so I set it to false.
Since sync_mode is not related to frame synchronization, I didn’t set it. But should I change it from standalone to false to make it work properly?

kazu-321, 2025-03-17:
Sorry, I didn't read it properly.
Is it enough to just change `stand_alone` to `free_run` to solve the issue? Or do I also need to set `enable_frame_sync` to false?
With Femto Bolt, I was able to update the FPS independently from depth without changing `sync_mode`.
Which behavior is correct?

jjiszjj, 2025-03-18:
@kazu-321 You can try changing `stand_alone` to `free_run`, which can solve the problem that Femto Mega cannot update the frame rate of depth and color independently.No need to set `enable_frame_sync` to false

kazu-321, 2025-03-23:
@jjiszjj 
Color FPS is now set correctly! Thank you very much!!!!
However, the depth FPS change is not applied.
I forked the repository and modified the launch file (as I thought it would make the changes easier to track).
Repository: https://github.com/kazu-321/OrbbecSDK_ROS2
Changed launch file: https://github.com/kazu-321/OrbbecSDK_ROS2/blob/main/orbbec_camera/launch/femto_mega.launch.py
Commit: https://github.com/kazu-321/OrbbecSDK_ROS2/commit/f1c7ad11d564cc462357066fff2250ab91404e8d

I saved the data using rosbag and checked the topic frequency with `mcap info`.
```
↪ $ mcap info rosbag2_2025_03_23-10_30_56/rosbag2_2025_03_23-10_30_56_0.mcap 
library:   libmcap 0.8.0                                               
profile:   ros2                                                        
messages:  1079                                                        
duration:  30.797576252s                                               
start:     2025-03-23T10:30:56.351213411+09:00 (1742693456.351213411)  
end:       2025-03-23T10:31:27.148789663+09:00 (1742693487.148789663)  
compression:
        : [598/598 chunks] [2.22 GiB/2.22 GiB (0.00%)] [73.86 MiB/sec] 
channels:
        (1) /camera/depth/image_raw  925 msgs (30.03 Hz)   : sensor_msgs/msg/Image [ros2msg]  
        (2) /camera/color/image_raw  154 msgs (5.00 Hz)    : sensor_msgs/msg/Image [ros2msg]  
channels: 2
attachments: 0
metadata: 1
```

kazu-321, 2025-03-23:
Also, I cannot select MJPG for 2K and 4K resolutions when using a LAN cable.
Why is that?

jjiszjj, 2025-03-25:
@kazu-321 Regarding the issue of depth FPS changes not being applied, you can check if `ir_fps` is also set to 5, and the resolution and frame rate of depth and ir should be consistent.https://github.com/orbbec/OrbbecSDK_ROS2/blob/cbd4331e44af9216496522e6fda881e112557665/orbbec_camera/launch/femto_mega.launch.py#L52

jjiszjj, 2025-03-25:
@kazu-321 I use LAN cable to connect Femto Mega in the lab and MJPG is effective in 2K and 4K resolution. The firmware version I use is [1.3.0](https://github.com/orbbec/OrbbecFirmware/releases/tag/Femto-Mega-Firmware). You can check whether the network bandwidth is sufficient. The run command I used for the test is:
`ros2 launch orbbec_camera femto_mega.launch.py sync_mode:=free_run net_device_ip:=192.168.1.10 net_device_port:=8090 color_fps:=5 depth_fps:=5 ir_fps:=5 color_width:=3840 color_height:=2160`

kazu-321, 2025-03-30:
[image]
I updated the firmware with the latest Orbbec viewer, but then updated it again with the version of the viewer specified in the manual.
[image]
The setup is going well so far. Thank you!
I'll continue testing so please keep it open for a little while longer.

kazu-321, 2025-03-30:
@jjiszjj 
I've created a PR.
Thanks a lot for the help!
https://github.com/orbbec/OrbbecSDK_ROS2/pull/104

kazu-321, 2025-03-30:
Please wait a little longer as the PR source was incorrect.