orbbec/OrbbecSDK_ROS2 #98: Femto mega device synchronization issue

anilbhujel, 2025-03-18:
Hello,
I have six Femto Mega cameras connected in network and I want to sync all the camera frames. I have Sync Hub Pro and necessary connection converters, but the cameras are not syncing at all. I had changed the camera syncrhonization configuration through OrbbecViewer (when I checked there, the secondary camera didn't record until the primary camera is not turned on, which seems the configuration is set) and also changed in launch file as shown in below snapshot. How can I synced the cameras? For example, I have shown here only for two cameras. 

[image]

[image]

[image]

[image]

anilbhujel, 2025-03-24:
Hi there,
any suggestion on this issue?

jjiszjj, 2025-03-25:
@anilbhujel The primary camera needs to set the synchronization signal output. You can add a `'trigger_out_enabled': 'true'` on camera_1.https://github.com/orbbec/OrbbecSDK_ROS2/blob/cbd4331e44af9216496522e6fda881e112557665/orbbec_camera/launch/femto_mega.launch.py#L94
You can also refer to our multi-machine synchronization example:https://github.com/orbbec/OrbbecSDK_ROS2/tree/v2-main/orbbec_camera/examples/multi_camera_synced

jjiszjj, 2025-03-25:
@anilbhujel The primary camera needs to set the synchronization signal output. You can add a 'trigger_out_enabled': 'true' on camera_1.https://github.com/orbbec/OrbbecSDK_ROS2/blob/cbd4331e44af9216496522e6fda881e112557665/orbbec_camera/launch/femto_mega.launch.py#L94

You can also refer to our multi-machine synchronization example:https://github.com/orbbec/OrbbecSDK_ROS2/tree/v2-main/orbbec_camera/examples/multi_camera_synced

anilbhujel, 2025-04-15:
Hi @jjiszjj,
Thank you for the information. Indeed neither works for me. Here is my detailed about testing, can you please suggest what I missed or shouldn't do to get the synced frame.
Hardware connection and setup:
1. I connected the two Femto Mega camera in the same private network with computer through a router and switch.
2. Assign computer IP: 192.168.0.101, Primary camera IP: 192.168.0.110, and the Secondary camera IP: 192.168.0.112.
3. Test the camera connection through network and found working correctly.
4. I used standard T568B ethernet cable to connect the devices with sync hub.

[image]
[image]

5. Configure the Primary and Secondary camera sync configuration following https://github.com/orbbec/Multi-Device-Synchronization-Example link. 
6. From the orbbecviewer, I noticed one thing that if we open the orbbecviewer by just clicking its icon the primary camera cannot enable Trigger Out Enable toggle switch (its like disable) but the secondary camera Trigger Out Enable switch always shows enable (even I  disabled and write it shows enable when I reconnect the camera)

[image]
I can see it if I used OrbbecViewer that I used to update the firmware with sudo ./OrbbecViewer command able to show Trigger Out Enable option 

[image]

7. However, as suggested in https://github.com/orbbec/Multi-Device-Synchronization-Example secondary camera operation controls by the Primary camera operation, only the frames are not getting with time synced.
8. I am using OrbbecSDK_ROS2 and changed this https://github.com/orbbec/OrbbecSDK_ROS2/blob/main/orbbec_camera/launch/multi_camera_synced.launch.py launch file as shown in below figure.

[image]

This is the femto_mega.launch.py default parameters

[image]

9. And then when I launch the camera, I saw the camera_1 (primary) launching at the last, but the timestamp is not syncing at all. However, the secondary camera is responding (it stopped) if I disconnect the sync cable from the Primary/secondary camera. 

[image]

10. When I followed the steps given in https://github.com/orbbec/Multi-Device-Synchronization-Example, I able to see the following view but I am not sure its synced or not and is not usable for me because I need ROS2 format.

[image]

 Can you please suggest me, how should I proceed? The cables connected to the sync hub are well tested in other operation like connecting cameras, so there might not be the cable issues.

anilbhujel, 2025-04-17:
Hi @jjiszjj 
Finally, it seems working. Thank your for your support.

[image]

closed 2025-04-17 by anilbhujel