orbbec/OrbbecSDK_ROS2 #99: GlContext::Initialize eglBindAPI error code = 0x300c on firefly ITX-3588J

vibealbert, 2025-03-19:
```
[component_container-1] [INFO] [1742374522.419462682] [camera.camera_container]: Load Library: /home/firefly/ros2_ws/install/orbbec_camera/lib/liborbbec_camera.so
[component_container-1] [INFO] [1742374522.560319184] [camera.camera_container]: Found class: rclcpp_components::NodeFactoryTemplate<orbbec_camera::OBCameraNodeDriver>
[component_container-1] [INFO] [1742374522.560391515] [camera.camera_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<orbbec_camera::OBCameraNodeDriver>
[component_container-1] [03/19 08:55:22.568821][info][58479][Context.cpp:68] Context created with config: /home/firefly/ros2_ws/install/orbbec_camera/share/orbbec_camera/config/OrbbecSDKConfig_v1.0.xml
[component_container-1] [03/19 08:55:22.568848][info][58479][Context.cpp:73] Work directory=/home/firefly/ros2_ws, SDK version=v1.10.18-20250103-27bf2f6
[component_container-1] [03/19 08:55:22.568868][info][58479][LinuxPal.cpp:32] createObPal: create LinuxPal!
[component_container-1] [03/19 08:55:22.767846][info][58479][LinuxPal.cpp:166] Create PollingDeviceWatcher!
[component_container-1] [03/19 08:55:22.768120][info][58479][DeviceManager.cpp:15] Current found device(s): (1)
[component_container-1] [03/19 08:55:22.768174][info][58479][DeviceManager.cpp:24] 	- Name: Femto Bolt, PID: 0x066b, SN/ID: CL8K14100SL, Connection: USB3.1
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/camera/camera' in container '/camera/camera_container'
[component_container-1] [INFO] [1742374522.874230855] [camera.camera]: Connecting to the default device
[component_container-1] [INFO] [1742374523.771010921] [camera.camera]: Select device cost 896 ms
[component_container-1] [INFO] [1742374523.771235789] [camera.camera]: Try to connect device via USB3.1
[component_container-1] [INFO] [1742374523.771520154] [camera.camera]: OBCameraNode: use_intra_process: OFF
[component_container-1] [INFO] [1742374523.789044250] [camera.camera]: current time domain: device
[component_container-1] [INFO] [1742374523.804844449] [camera.camera]: Setting heartbeat to OFF
[component_container-1] [INFO] [1742374523.806299814] [camera.camera]: Current sync mode: OB_MULTI_DEVICE_SYNC_MODE_STANDALONE
[component_container-1] [INFO] [1742374523.808210748] [camera.camera]: Set sync mode: OB_MULTI_DEVICE_SYNC_MODE_STANDALONE
[component_container-1] [INFO] [1742374523.808330618] [camera.camera]: Setting depth soft filter to ON
[component_container-1] [INFO] [1742374523.808493363] [camera.camera]: Setting color auto white balance to ON
[component_container-1] [INFO] [1742374523.813859550] [camera.camera]: Setting color auto exposure to ON
[component_container-1] [INFO] [1742374524.155211004] [camera.camera]: default_soft_filter_max_diff: 300
[component_container-1] [INFO] [1742374524.155606199] [camera.camera]: default_soft_filter_speckle_size: 25
[component_container-1] [INFO] [1742374524.162148928] [camera.camera]:  stream color is enabled - width: 1280, height: 960, fps: 5, Format: OB_FORMAT_MJPG
[component_container-1] [INFO] [1742374524.164636466] [camera.camera]:  stream depth is enabled - width: 640, height: 576, fps: 5, Format: OB_FORMAT_Y16
[component_container-1] [INFO] [1742374524.167695651] [camera.camera]:  stream ir is enabled - width: 640, height: 576, fps: 5, Format: OB_FORMAT_Y16
[component_container-1] [INFO] [1742374524.168103970] [camera.camera]: stream gyro full scale range 1000dps sample rate 200hz
[component_container-1] [INFO] [1742374524.168409626] [camera.camera]: stream accel full scale range 4g sample rate 200hz
[component_container-1] [INFO] [1742374524.348097515] [camera.camera]: Publish diagnostics every 1 seconds
[component_container-1] [INFO] [1742374524.513409323] [camera.camera]: start accel stream with range: 4g,rate:200hz, and start gyro stream with range:1000dps,rate:200hz
[component_container-1] [INFO] [1742374524.513785852] [camera.camera]: Enable color stream
[component_container-1] [INFO] [1742374524.513875682] [camera.camera]: Stream color width: 1280 height: 960 fps: 5 format: MJPG
[component_container-1] [INFO] [1742374524.513941013] [camera.camera]: Enable depth stream
[component_container-1] [INFO] [1742374524.514004011] [camera.camera]: Stream depth width: 640 height: 576 fps: 5 format: Y16
[component_container-1] [INFO] [1742374524.514064092] [camera.camera]: Enable ir stream
[component_container-1] [INFO] [1742374524.514125632] [camera.camera]: Stream ir width: 640 height: 576 fps: 5 format: Y16
[component_container-1] arm_release_ver: g13p0-01eac0, rk_so_ver: 9
[component_container-1] GlContext::Initialize eglBindAPI error code = 0x300c
[component_container-1] GlContext::Initialize eglBindAPI error code = 0x300c
[component_container-1] GlContext::Initialize eglBindAPI error code = 0x300c
[component_container-1] GlContext::Initialize eglBindAPI error code = 0x300c
[component_container-1] GlContext::Initialize eglBindAPI error code = 0x300c
[component_container-1] GlContext::Initialize eglBindAPI error code = 0x300c
[component_container-1] GlContext::Initialize eglBindAPI error code = 0x300c
[component_container-1] GlContext::Initialize eglBindAPI error code = 0x300c
[component_container-1] GlContext::Initialize eglBindAPI error code = 0x300c
[component_container-1] GlContext::Initialize eglBindAPI error code = 0x300c
[component_container-1] [INFO] [1742374526.159330503] [camera.camera]: Enable frame sync
[component_container-1] [INFO] [1742374526.162375105] [camera.camera]: Device Femto Bolt connected
[component_container-1] [INFO] [1742374526.162438103] [camera.camera]: Serial number: CL8K14100SL
[component_container-1] [INFO] [1742374526.162481560] [camera.camera]: Firmware version: 1.0.9
[component_container-1] [INFO] [1742374526.162522683] [camera.camera]: Hardware version: 1.2
[component_container-1] [INFO] [1742374526.162556224] [camera.camera]: device unique id: 4-1.4-6
[component_container-1] [INFO] [1742374526.162594139] [camera.camera]: Current node pid: 58479
[component_container-1] [INFO] [1742374526.162626805] [camera.camera]: usb connect type: USB3.1
[component_container-1] [INFO] [1742374526.162659762] [camera.camera]: Start device cost 3388 ms
[component_container-1] [INFO] [1742374526.162692136] [camera.camera]: Initialize device cost 2391 ms
[component_container-1] [INFO] [1742374527.669319342] [camera.camera]: Publishing static transform from ir to depth
[component_container-1] [INFO] [1742374527.669707245] [camera.camera]: Translation 0, 0, 0
[component_container-1] [INFO] [1742374527.669911696] [camera.camera]: Rotation 0, 0, 0, 1
[component_container-1] [INFO] [1742374527.670108273] [camera.camera]: Publishing static transform from color to depth
[component_container-1] [INFO] [1742374527.670360556] [camera.camera]: Translation 32.4789, 1.43571, -1.71275
[component_container-1] [INFO] [1742374527.670569673] [camera.camera]: Rotation -0.00103901, -0.0534426, 0.000558116, 0.997858
[component_container-1] [INFO] [1742374527.670755459] [camera.camera]: Publishing static transform from depth to depth
[component_container-1] [INFO] [1742374527.671202276] [camera.camera]: Translation 0, 0, 0
[component_container-1] [INFO] [1742374527.671410519] [camera.camera]: Rotation 0, 0, 0, 1
[component_container-1] [INFO] [1742374527.671598637] [camera.camera]: Publishing static transform from accel to depth
[component_container-1] [INFO] [1742374527.671796964] [camera.camera]: Translation -37.8292, -1.29141, 54.174
[component_container-1] [INFO] [1742374527.671989457] [camera.camera]: Rotation 0.525483, 0.473147, -0.473147, 0.525483
[component_container-1] [INFO] [1742374527.672183992] [camera.camera]: Publishing static transform from gyro to depth
[component_container-1] [INFO] [1742374527.672379985] [camera.camera]: Translation -37.8292, -1.29141, 54.174
[component_container-1] [INFO] [1742374527.672574228] [camera.camera]: Rotation 0.525483, 0.473147, -0.473147, 0.525483
```

jjiszjj, 2025-04-03:
@vibealbert Hi.Thank you very much for your feedback.I think it may be that your current system lacks EGL support.You can try running`sudo apt install libegl1 libegl-dev`

vibealbert, 2025-04-08:
> [@vibealbert](https://github.com/vibealbert) Hi.Thank you very much for your feedback.I think it may be that your current system lacks EGL support.You can try running`sudo apt install libegl1 libegl-dev`
hi @jjiszjj, I tried the solution above, but it still not work, Is there any other way?
```
firefly@firefly:~/ros2_ws$ sudo apt list | grep libegl

libegl-amber0/jammy 21.3.7-0ubuntu1 arm64
libegl-dev/jammy,now 1.4.0-1 arm64 [installed]
libegl-mali-xlnx/jammy 9p0.01rel0-0ubuntu5 arm64
libegl-mesa0/jammy 22.0.1-1ubuntu2 arm64
libegl1-mesa-dev/jammy,now 22.0.1-1ubuntu2 arm64 [installed]
libegl1-mesa/jammy 22.0.1-1ubuntu2 arm64
libegl1/jammy,now 1.4.0-1 arm64 [installed]
```
this is the crash stack trace
```
Received signal: 6
Stack trace (most recent call last) in thread 2761:
#14   Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
#13   Object "/lib/aarch64-linux-gnu/libc.so.6", at 0x7f83ed5edb, in 
#12   Object "/lib/aarch64-linux-gnu/libc.so.6", at 0x7f83e6d5c7, in 
#11   Object "/lib/aarch64-linux-gnu/libstdc++.so.6", at 0x7f840a224b, in 
#10   Object "/home/firefly/ros2_ws/install/orbbec_camera/lib/libOrbbecSDK.so.2", at 0x7f7d6a2977, in 
#9    Object "/home/firefly/ros2_ws/install/orbbec_camera/lib/libOrbbecSDK.so.2", at 0x7f7d69ac8f, in 
#8    Object "/lib/aarch64-linux-gnu/libstdc++.so.6", at 0x7f84072da3, in __cxa_throw
#7    Object "/lib/aarch64-linux-gnu/libstdc++.so.6", at 0x7f84072abf, in std::terminate()
#6    Object "/lib/aarch64-linux-gnu/libstdc++.so.6", at 0x7f84072a5b, in 
#5    Object "/lib/aarch64-linux-gnu/libstdc++.so.6", at 0x7f8407527b, in __gnu_cxx::__verbose_terminate_handler()
#4    Object "/lib/aarch64-linux-gnu/libc.so.6", at 0x7f83e1712f, in abort
#3    Object "/lib/aarch64-linux-gnu/libc.so.6", at 0x7f83e2a67b, in raise
#2    Object "/lib/aarch64-linux-gnu/libc.so.6", at 0x7f83e6f200, in 
#1    Object "linux-vdso.so.1", at 0x7f844b97af, in 
#0    Object "/home/firefly/ros2_ws/install/orbbec_camera/lib/liborbbec_camera.so", at 0x7f7ddea3f7, in signalHandler(int)
Received signal: 11
```