orbbec/OrbbecSDK_ROS2 #100: FEMTO Mega pointcloud data preprocessing

KingDo-c, 2025-03-21:
I'm using a femto Mega model, and I'm looking for a way to remove smoothing (curved) from the point cloud data I got. I didn't apply any filters with the launch option, but I couldn't solve it. Can you help?
The data cannot be distinguished because there is an interpolation between the object and the object.

jjiszjj, 2025-04-03:
@KingDo-c Hi,The Femto Mega is able to use the noise_removal_filter, you can add these three lines to femto_mega.launch.py.And set `enable_noise_removal_filter` to `true`

https://github.com/orbbec/OrbbecSDK_ROS2/blob/40cf9531e794ebbe89dc8709b02426c41f322584/orbbec_camera/launch/gemini_330_series.launch.py#L163C32-L163C59
https://github.com/orbbec/OrbbecSDK_ROS2/blob/40cf9531e794ebbe89dc8709b02426c41f322584/orbbec_camera/launch/gemini_330_series.launch.py#L171
https://github.com/orbbec/OrbbecSDK_ROS2/blob/40cf9531e794ebbe89dc8709b02426c41f322584/orbbec_camera/launch/gemini_330_series.launch.py#L172

jjiszjj, 2025-04-03:
@KingDo-c  Hi,The Femto Mega is able to use the noise_removal_filter, you can add these three lines to femto_mega.launch.py.And set `enable_noise_removal_filter` to `true`

https://github.com/orbbec/OrbbecSDK_ROS2/blob/40cf9531e794ebbe89dc8709b02426c41f322584/orbbec_camera/launch/gemini_330_series.launch.py#L163

https://github.com/orbbec/OrbbecSDK_ROS2/blob/40cf9531e794ebbe89dc8709b02426c41f322584/orbbec_camera/launch/gemini_330_series.launch.py#L171

https://github.com/orbbec/OrbbecSDK_ROS2/blob/40cf9531e794ebbe89dc8709b02426c41f322584/orbbec_camera/launch/gemini_330_series.launch.py#L172

jjiszjj, 2025-04-03:
@KingDo-c We will add `enable_noise_removal_filter`, `noise_removal_filter_min_diff` and `noise_removal_filter_max_size` to femto_mega.launch.py ​​in the next version.