orbbec/OrbbecSDK_ROS2 #102: bloom release

christian-rauch, 2025-03-27:
Can you please release this package via bloom for `jazzy` and `rolling` to the official ROS apt repo? See https://docs.ros.org/en/jazzy/How-To-Guides/Releasing/Releasing-a-Package.html for details.

jjiszjj, 2025-04-03:
@christian-rauch Hello, Thank you very much for your feedback. I will review the relevant documentation in the near future.

winf, 2025-08-12:
> [@jjiszjj](https://github.com/jjiszjj) Any updates on this? The documentation on how to release packages to the ROS repo is relative straight forward. See for example how the RealSense packages, the library and the node, are released.
Your suggestion is extremely helpful, many thanks. This repo has two significant branches: main and v2-main. Since some legacy devices only support the main branch, pushing only the v2-main branch to the public ROS 2 buildfarm would cause confusion for these users. After careful consideration, we have decided to provide downloadable deb packages in the next release cycle.

winf, 2025-08-21:
> [@jjiszjj](https://github.com/jjiszjj) [@winf](https://github.com/winf) Can you comment on, if you would still consider releasing the newest version to bloom?
> 
> This is what Intel is doing with [librealsense](https://github.com/IntelRealSense/librealsense). They are releasing the new version 2 as bloom package [`librealsense2`](https://index.ros.org/p/librealsense2/) so that the library (`sudo apt install ros-jazzy-librealsense2`) and camera nodes (`sudo apt install ros-jazzy-realsense2-description ros-jazzy-realsense2-camera`) are available as binary packages. The legacy version is still available under the v2 tags. This way, costumers of legacy devices can use the old version until they replace their hardware and customers with newer hardware can conveniently install this from the apt repo.
> 
> I would be happy to help a bit in setting this up.
We are already working on the release process, https://github.com/ros/rosdistro/pull/47308

christian-rauch, 2025-08-21:
> We are already working on the release process, [ros/rosdistro#47308](https://github.com/ros/rosdistro/pull/47308)

Thanks. I did not see this.

Are you also considering releasing this to `jazzy` and `rolling`?

ob-yalian, 2025-10-31:
@christian-rauch Since the main branch and v2-main branch use the same package names, they cannot both be released to the APT repository. After internal discussion, we decided to stop maintaining the main branch deb packages and will only maintain the v2-main branch.

Currently, we are waiting for the official review of our PR. Once it is approved, we will immediately upload the packages from the v2-main branch.https://github.com/ros/rosdistro/pull/48403