orbbec/OrbbecSDK_ROS2 #103: accel and gyro frame outside of camera

christian-rauch, 2025-03-28:
The frames `camera_accel_optical_frame` and `camera_gyro_optical_frame` are outside of the camera housing:

[image]

[image]

jjiszjj, 2025-04-03:
@christian-rauch Hello,Thank you very much for your feedback.There is indeed a problem with Femto_bolt in orbbec_description. We are considering whether to delete it or replace it with a new model. And We will also check whether camera_accel_optical_frame and camera_gyro_optical_frame are normal.

christian-rauch, 2025-09-01:
It looks like this issue is not present on the `v2-main` branch.