orbbec/OrbbecSDK_ROS2 #105: camera_accel_gyro_optical_frame is not published in tf

songWell, 2025-04-08:
Hello, I use the femto mega camera (ros2 launch orbbec_camera femto_mega.launch.py ​​depth_registration:=true), and I use the output of imu (/camera/gyro_accel/sample), but the frame_id camera_accel_gyro_optical_frame of this message is not published in tf, am I missing any settings?

jjiszjj, 2025-04-14:
@songWell In OrbbecSDK_ROS2, `camera_accel_gyro_optical_frame` will not be published in tf, but `camera_gyro_optical_frame` and `camera_accel_optical_frame` will be published in tf

songWell, 2025-04-15:
Thank you for your answer, so _camera_accel_gyro_optical_frame_ is consistent with _camera_gyro_optical_frame_ and _camera_accel_optical_frame_.

closed 2025-04-15 by songWell