orbbec/OrbbecSDK_ROS2 #107: Failed to initialize device uvc_open failed: [Path: 2-3.1-7.0, Return Code: -6]

retazo0018, 2025-04-23:
Hi @jian-dong, @Kashery,

I have a Gemini 335L camera. I am trying to install the setup inside ISAAC-ROS Dev container in Jetson AGX Orin. 

When I try to Install udev rules as mentioned in README.md, the following output appears. I do not get this message when I run the same commands outside the docker container.
[image]

The colcon build is successful and when I try to launch the camera by `ros2 launch orbbec_camera gemini_330_series.launch.py`, I get the following error.
[image]

Please suggest some pointers to resolve this issue.

Thanks,

jjiszjj, 2025-04-24:
@retazo0018 Hi,I found that you are starting the camera in docker, you can try adding these two parameters `--privileged` and `--device=/dev` when running docker

retazo0018, 2025-04-24:
Hi @jjiszjj ,

Thanks for your reply. I included both in my docker run command, I still have the same error.

Is there a recommended way to launch the camera in the [ISAAC-ROS dev](https://nvidia-isaac-ros.github.io/getting_started/dev_env_setup.html) docker environment?

closed 2025-11-21 by ob-yalian