orbbec/OrbbecSDK_ROS2 #110: How to use depth_precicion pram in G330 launch file

cfsdp, 2025-04-25:
Due to my project required, I need more accurate depth value. I tried set DepthUnit to 0.1 in OrbbecView and it worked well. But when I try it in ROS2 Wrapper, I seems not working. In my Ros program the depth still output 0.400000(m) or 0.399000(m) e.g.
The log is :
My  launch command is "ros2 launch orbbec_camera gemini_330_series.launch.py depth_precision:=0.1mm"
[component_container-1] [INFO] [1745554610.417454073] [camera.camera_container]: Load Library: /home/gao/oil-yp-robot/install/orbbec_camera/lib/liborbbec_camera.so
[component_container-1] [INFO] [1745554610.492415988] [camera.camera_container]: Found class: rclcpp_components::NodeFactoryTemplate<orbbec_camera::OBCameraNodeDriver>
[component_container-1] [INFO] [1745554610.492466235] [camera.camera_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<orbbec_camera::OBCameraNodeDriver>
[component_container-1] [INFO] [1745554610.508205328] [camera.camera]: setUvcBackendType:libuvc
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/camera/camera' in container '/camera/camera_container'
[component_container-1] [INFO] [1745554612.588751326] [camera.camera]: Connecting to the default device
[component_container-1] [INFO] [1745554612.750260453] [camera.camera]: Select device cost 161 ms
[component_container-1] [INFO] [1745554612.750301491] [camera.camera]: Try to connect device via USB3.2
[component_container-1] [INFO] [1745554612.750350775] [camera.camera]: OBCameraNode: use_intra_process: OFF
[component_container-1] [INFO] [1745554612.752182138] [camera.camera]: current time domain: global
[component_container-1] [INFO] [1745554612.752197683] [camera.camera]: hdr_index1_laser_control_ 1 hdr_index1_depth_exposure_ 1 hdr_index1_depth_gain_ 16 hdr_index1_ir_brightness_ 20 hdr_index1_ir_ae_max_exposure_ 2000
[component_container-1] 
[component_container-1] [INFO] [1745554612.752202781] [camera.camera]: hdr_index0_laser_control_ 1 hdr_index0_depth_exposure_ 7500 hdr_index0_depth_gain_ 16 hdr_index0_ir_brightness_ 60 hdr_index0_ir_ae_max_exposure_ 10000
[component_container-1] 
[component_container-1] [INFO] [1745554612.752207530] [camera.camera]: laser_index1_laser_control_ 0 laser_index1_depth_exposure_ 3000 laser_index1_depth_gain_ 16 laser_index1_ir_brightness_ 60 laser_index1_ir_ae_max_exposure_ 17000
[component_container-1] 
[component_container-1] [INFO] [1745554612.752213082] [camera.camera]: laser_index0_laser_control_ 1 laser_index0_depth_exposure_ 3000 laser_index0_depth_gain_ 16 laser_index0_ir_brightness_ 60 laser_index0_ir_ae_max_exposure_ 30000
[component_container-1] 
[component_container-1] [INFO] [1745554612.873771028] [camera.camera]: Setting heartbeat to OFF
[component_container-1] [INFO] [1745554612.873817644] [camera.camera]: Create align filter
[component_container-1] [INFO] [1745554612.874202816] [camera.camera]: Depth process is HW D2D
[component_container-1] [INFO] [1745554612.874207283] [camera.camera]: Setting LDP to ON
[component_container-1] [INFO] [1745554612.875218168] [camera.camera]: Setting G300 laser control to 1
[component_container-1] [INFO] [1745554612.875350573] [camera.camera]: Available presets:
[component_container-1] [INFO] [1745554612.875360564] [camera.camera]: Preset 0: Default
[component_container-1] [INFO] [1745554612.875363140] [camera.camera]: Preset 1: Hand
[component_container-1] [INFO] [1745554612.875365740] [camera.camera]: Preset 2: High Accuracy
[component_container-1] [INFO] [1745554612.875368082] [camera.camera]: Preset 3: High Density
[component_container-1] [INFO] [1745554612.875370267] [camera.camera]: Preset 4: Medium Density
[component_container-1] [INFO] [1745554612.875372774] [camera.camera]: Preset 5: Factory Calib
[component_container-1] [INFO] [1745554612.875375067] [camera.camera]: Preset 6: Custom
[component_container-1] [INFO] [1745554612.875377378] [camera.camera]: Load device preset: Default
[component_container-1] [INFO] [1745554612.876673791] [camera.camera]: Device preset Default loaded
[component_container-1] [INFO] [1745554612.876836766] [camera.camera]: Current sync mode: OB_MULTI_DEVICE_SYNC_MODE_STANDALONE
[component_container-1] [INFO] [1745554612.877306909] [camera.camera]: Set sync mode: OB_MULTI_DEVICE_SYNC_MODE_STANDALONE
[component_container-1] [INFO] [1745554612.877314970] [camera.camera]: enable_noise_removal_filter:1
[component_container-1] [INFO] [1745554612.877318080] [camera.camera]: Setting color auto white balance to ON
[component_container-1] [INFO] [1745554612.877480944] [camera.camera]: Setting color auto exposure priority to OFF
[component_container-1] [INFO] [1745554612.877548304] [camera.camera]: Setting color auto exposure to ON
[component_container-1] [INFO] [1745554612.877747864] [camera.camera]: Setting color backlight compenstation to OFF
[component_container-1] [INFO] [1745554612.877909702] [camera.camera]: Setting depth auto exposure priority to OFF
[component_container-1] [INFO] [1745554612.878060075] [camera.camera]: Setting IR auto exposure to ON
[component_container-1] [INFO] [1745554612.880497302] [camera.camera]: default noise removal filter min diff: 256
[component_container-1] [INFO] [1745554612.880521612] [camera.camera]: default noise removal filter max size: 80
[component_container-1] [INFO] [1745554612.880765465] [camera.camera]: Setting hardware noise removal filter:0
[component_container-1] [INFO] [1745554612.880917110] [camera.camera]: Setting DecimationFilter......
[component_container-1] [INFO] [1745554612.880936162] [camera.camera]: set DecimationFilter to false
[component_container-1] [INFO] [1745554612.880954911] [camera.camera]: Skip setting filter: DecimationFilter
[component_container-1] [INFO] [1745554612.880965069] [camera.camera]: Setting HDRMerge......
[component_container-1] [INFO] [1745554612.880999964] [camera.camera]: set HDRMerge to false
[component_container-1] [INFO] [1745554612.881015298] [camera.camera]: Skip setting filter: HDRMerge
[component_container-1] [INFO] [1745554612.881024617] [camera.camera]: Setting SequenceIdFilter......
[component_container-1] [INFO] [1745554612.881032305] [camera.camera]: set SequenceIdFilter to false
[component_container-1] [INFO] [1745554612.881039955] [camera.camera]: Skip setting filter: SequenceIdFilter
[component_container-1] [INFO] [1745554612.881048287] [camera.camera]: Setting SpatialAdvancedFilter......
[component_container-1] [INFO] [1745554612.881055832] [camera.camera]: set SpatialAdvancedFilter to false
[component_container-1] [INFO] [1745554612.881063635] [camera.camera]: Skip setting filter: SpatialAdvancedFilter
[component_container-1] [INFO] [1745554612.881071767] [camera.camera]: Setting TemporalFilter......
[component_container-1] [INFO] [1745554612.881079217] [camera.camera]: set TemporalFilter to false
[component_container-1] [INFO] [1745554612.881086818] [camera.camera]: Skip setting filter: TemporalFilter
[component_container-1] [INFO] [1745554612.881094843] [camera.camera]: Setting HoleFillingFilter......
[component_container-1] [INFO] [1745554612.881102174] [camera.camera]: set HoleFillingFilter to false
[component_container-1] [INFO] [1745554612.881110013] [camera.camera]: Skip setting filter: HoleFillingFilter
[component_container-1] [INFO] [1745554612.881118826] [camera.camera]: Setting DisparityTransform......
[component_container-1] [INFO] [1745554612.881126382] [camera.camera]: Skip setting filter: DisparityTransform
[component_container-1] [INFO] [1745554612.881134208] [camera.camera]: Setting ThresholdFilter......
[component_container-1] [INFO] [1745554612.881141650] [camera.camera]: set ThresholdFilter to false
[... 13 lines cut ...]
[component_container-1] [INFO] [1745554612.915304108] [camera.camera]: Publish diagnostics every 1 seconds
[component_container-1] [INFO] [1745554612.915885945] [camera.camera]: Enable color stream
[component_container-1] [INFO] [1745554612.915899872] [camera.camera]: color video_profile: 1280x720 30fps OB_FORMAT_MJPG
[component_container-1] [INFO] [1745554612.915906281] [camera.camera]: Stream color width: 1280 height: 720 fps: 30 format: MJPG
[component_container-1] [INFO] [1745554612.915911308] [camera.camera]: Enable depth stream
[component_container-1] [INFO] [1745554612.915914930] [camera.camera]: Stream depth width: 848 height: 480 fps: 30 format: Y16
[component_container-1] [INFO] [1745554612.931347091] [camera.camera]: Enable frame sync
[... cut ...]
[component_container-1] [INFO] [1745554613.982914408] [camera.camera]: Hardware version: 0.1
[component_container-1] [INFO] [1745554613.982923011] [camera.camera]: device unique id: 2-1-9
[component_container-1] [INFO] [1745554613.982935786] [camera.camera]: Current node pid: 418064
[component_container-1] [INFO] [1745554613.982944949] [camera.camera]: usb connect type: USB3.2
[component_container-1] [INFO] [1745554613.982953654] [camera.camera]: Start device cost 1404 ms
[component_container-1] [INFO] [1745554613.982962794] [camera.camera]: Initialize device cost 1232 ms

jjiszjj, 2025-05-07:
@cfsdp Hello,Thank you very much for your feedback.You can modify this parameter to use depth_precision
https://github.com/orbbec/OrbbecSDK_ROS2/blob/0eb7f445265a0bb8f1d74df913c698244f02707e/orbbec_camera/launch/gemini_330_series.launch.py#L105

jjiszjj, 2025-05-07:
@cfsdp  Hello,Thank you very much for your feedback.We recently updated the use of the depth_precision parameter.You can modify this parameter to use depth_precision
https://github.com/orbbec/OrbbecSDK_ROS2/blob/0eb7f445265a0bb8f1d74df913c698244f02707e/orbbec_camera/launch/gemini_330_series.launch.py#L207

For example: 
`ros2 launch orbbec_camera gemini_330_series.launch.py depth_precision:=0.1mm`