orbbec/OrbbecSDK_ROS2 #114: Femto Bolt "NoiseRemovalFilter#1 doesn't exist" on v2

christian-rauch, 2025-05-23:
I successfully ran `ros2 launch orbbec_camera femto_bolt.launch.py` on the `main` branch. But on `v2-main` this fails to start and loops forever:
```
[component_container-1] [INFO] [1748012165.019914818] [camera.camera_container]: Load Library: /home/christian/dev/orbbec_ws/install/orbbec_camera/lib/liborbbec_camera.so
[component_container-1] [INFO] [1748012165.057334536] [camera.camera_container]: Found class: rclcpp_components::NodeFactoryTemplate<orbbec_camera::OBCameraNodeDriver>
[component_container-1] [INFO] [1748012165.057355498] [camera.camera_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<orbbec_camera::OBCameraNodeDriver>
[component_container-1] [INFO] [1748012165.060824212] [camera.camera]: setUvcBackendType:libuvc
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/camera/camera' in container '/camera/camera_container'
[component_container-1] [INFO] [1748012166.253674526] [camera.camera]: Connecting to the default device
[component_container-1] [INFO] [1748012166.265998306] [camera.camera]: Select device cost 12 ms
[component_container-1] [INFO] [1748012166.266141804] [camera.camera]: Try to connect device via USB3.1
[component_container-1] [INFO] [1748012166.266369115] [camera.camera]: OBCameraNode: use_intra_process: OFF
[component_container-1] [INFO] [1748012166.428781600] [camera.camera]: current time domain: global
[component_container-1] [INFO] [1748012166.428866328] [camera.camera]: hdr_index1_laser_control_ 1 hdr_index1_depth_exposure_ 1 hdr_index1_depth_gain_ 16 hdr_index1_ir_brightness_ 20 hdr_index1_ir_ae_max_exposure_ 2000
[component_container-1] 
[component_container-1] [INFO] [1748012166.428884844] [camera.camera]: hdr_index0_laser_control_ 1 hdr_index0_depth_exposure_ 7500 hdr_index0_depth_gain_ 16 hdr_index0_ir_brightness_ 60 hdr_index0_ir_ae_max_exposure_ 10000
[component_container-1] 
[component_container-1] [INFO] [1748012166.428898518] [camera.camera]: laser_index1_laser_control_ 0 laser_index1_depth_exposure_ 3000 laser_index1_depth_gain_ 16 laser_index1_ir_brightness_ 60 laser_index1_ir_ae_max_exposure_ 7000
[component_container-1] 
[component_container-1] [INFO] [1748012166.428914082] [camera.camera]: laser_index0_laser_control_ 1 laser_index0_depth_exposure_ 3000 laser_index0_depth_gain_ 16 laser_index0_ir_brightness_ 60 laser_index0_ir_ae_max_exposure_ 17000
[component_container-1] 
[component_container-1] [INFO] [1748012167.314123672] [camera.camera]: Setting heartbeat to OFF
[component_container-1] [INFO] [1748012167.314698547] [camera.camera]: Current sync mode: OB_MULTI_DEVICE_SYNC_MODE_STANDALONE
[component_container-1] [INFO] [1748012167.315668377] [camera.camera]: Set sync mode: OB_MULTI_DEVICE_SYNC_MODE_STANDALONE
[component_container-1] [INFO] [1748012167.315701557] [camera.camera]: Setting color mirror to OFF
[component_container-1] [INFO] [1748012167.315762545] [camera.camera]: Setting color flip to OFF
[component_container-1] [INFO] [1748012167.315779852] [camera.camera]: Setting depth mirror to OFF
[component_container-1] [INFO] [1748012167.315812454] [camera.camera]: Setting depth flip to OFF
[component_container-1] [INFO] [1748012167.315827816] [camera.camera]: Setting ir mirror to OFF
[component_container-1] [INFO] [1748012167.315855747] [camera.camera]: Setting ir flip to OFF
[component_container-1] [INFO] [1748012167.315915607] [camera.camera]: Setting noise removal filter:OFF
[component_container-1] [INFO] [1748012167.315927464] [camera.camera]: Setting color auto white balance to ON
[component_container-1] [INFO] [1748012167.316136127] [camera.camera]: Setting color auto exposure to ON
[component_container-1] [ERROR] [1748012167.316617876] [camera.camera]: Failed to setup topics: Filter@FrameProcessor: config item NoiseRemovalFilter#1 doesn't exist
[component_container-1] [ERROR] [1748012167.316798783] [camera.camera]: Failed to initialize device (Attempt 1 of 3): Filter@FrameProcessor: config item NoiseRemovalFilter#1 doesn't exist
[component_container-1] [INFO] [1748012167.316912448] [camera.camera]: OBCameraNode: use_intra_process: OFF
[component_container-1] [INFO] [1748012167.317384058] [camera.camera]: current time domain: global
[component_container-1] [INFO] [1748012167.317403260] [camera.camera]: hdr_index1_laser_control_ 1 hdr_index1_depth_exposure_ 1 hdr_index1_depth_gain_ 16 hdr_index1_ir_brightness_ 20 hdr_index1_ir_ae_max_exposure_ 2000
[component_container-1] 
[component_container-1] [INFO] [1748012167.317416170] [camera.camera]: hdr_index0_laser_control_ 1 hdr_index0_depth_exposure_ 7500 hdr_index0_depth_gain_ 16 hdr_index0_ir_brightness_ 60 hdr_index0_ir_ae_max_exposure_ 10000
[component_container-1] 
[component_container-1] [INFO] [1748012167.317427320] [camera.camera]: laser_index1_laser_control_ 0 laser_index1_depth_exposure_ 3000 laser_index1_depth_gain_ 16 laser_index1_ir_brightness_ 60 laser_index1_ir_ae_max_exposure_ 7000
[component_container-1] 
[component_container-1] [INFO] [1748012167.317439508] [camera.camera]: laser_index0_laser_control_ 1 laser_index0_depth_exposure_ 3000 laser_index0_depth_gain_ 16 laser_index0_ir_brightness_ 60 laser_index0_ir_ae_max_exposure_ 17000
[component_container-1] 
[component_container-1] [INFO] [1748012167.319343388] [camera.camera]: Setting heartbeat to OFF
[component_container-1] [INFO] [1748012167.319442866] [camera.camera]: Current sync mode: OB_MULTI_DEVICE_SYNC_MODE_STANDALONE
[component_container-1] [INFO] [1748012167.319463948] [camera.camera]: Set sync mode: OB_MULTI_DEVICE_SYNC_MODE_STANDALONE
[component_container-1] [INFO] [1748012167.319484924] [camera.camera]: Setting color mirror to OFF
[component_container-1] [INFO] [1748012167.319508571] [camera.camera]: Setting color flip to OFF
[component_container-1] [INFO] [1748012167.319526449] [camera.camera]: Setting depth mirror to OFF
[component_container-1] [INFO] [1748012167.319543378] [camera.camera]: Setting depth flip to OFF
[component_container-1] [INFO] [1748012167.319559530] [camera.camera]: Setting ir mirror to OFF
[component_container-1] [INFO] [1748012167.319577540] [camera.camera]: Setting ir flip to OFF
[component_container-1] [INFO] [1748012167.319634589] [camera.camera]: Setting noise removal filter:OFF
[component_container-1] [INFO] [1748012167.319648605] [camera.camera]: Setting color auto white balance to ON
[component_container-1] [INFO] [1748012167.319865122] [camera.camera]: Setting color auto exposure to ON
[component_container-1] [ERROR] [1748012167.320298810] [camera.camera]: Failed to setup topics: Filter@FrameProcessor: config item NoiseRemovalFilter#1 doesn't exist
[component_container-1] [ERROR] [1748012167.320439134] [camera.camera]: Failed to initialize device (Attempt 2 of 3): Filter@FrameProcessor: config item NoiseRemovalFilter#1 doesn't exist
[component_container-1] [INFO] [1748012167.320571193] [camera.camera]: OBCameraNode: use_intra_process: OFF
[component_container-1] [INFO] [1748012167.321131875] [camera.camera]: current time domain: global
[component_container-1] [INFO] [1748012167.321154407] [camera.camera]: hdr_index1_laser_control_ 1 hdr_index1_depth_exposure_ 1 hdr_index1_depth_gain_ 16 hdr_index1_ir_brightness_ 20 hdr_index1_ir_ae_max_exposure_ 2000
[component_container-1] 
[component_container-1] [INFO] [1748012167.321170250] [camera.camera]: hdr_index0_laser_control_ 1 hdr_index0_depth_exposure_ 7500 hdr_index0_depth_gain_ 16 hdr_index0_ir_brightness_ 60 hdr_index0_ir_ae_max_exposure_ 10000
[component_container-1] 
[component_container-1] [INFO] [1748012167.321184864] [camera.camera]: laser_index1_laser_control_ 0 laser_index1_depth_exposure_ 3000 laser_index1_depth_gain_ 16 laser_index1_ir_brightness_ 60 laser_index1_ir_ae_max_exposure_ 7000
[component_container-1] 
[component_container-1] [INFO] [1748012167.321198437] [camera.camera]: laser_index0_laser_control_ 1 laser_index0_depth_exposure_ 3000 laser_index0_depth_gain_ 16 laser_index0_ir_brightness_ 60 laser_index0_ir_ae_max_exposure_ 17000
[component_container-1] 
[component_container-1] [INFO] [1748012167.323117847] [camera.camera]: Setting heartbeat to OFF
[component_container-1] [INFO] [1748012167.323184652] [camera.camera]: Current sync mode: OB_MULTI_DEVICE_SYNC_MODE_STANDALONE
[... 121 lines cut ...]
[INFO] [1748012619.405082093] [orbbec_camera_node]: setUvcBackendType:libuvc
[INFO] [1748012620.621738093] [orbbec_camera_node]: Connecting to the default device
[... cut ...]
[INFO] [1748012621.678664176] [orbbec_camera_node]: Setting color auto white balance to ON
[INFO] [1748012621.678853699] [orbbec_camera_node]: Setting color auto exposure to ON
[ERROR] [1748012621.679388435] [orbbec_camera_node]: Failed to setup topics: Filter@FrameProcessor: config item NoiseRemovalFilter#1 doesn't exist
[ERROR] [1748012621.679595885] [orbbec_camera_node]: Failed to initialize device (Attempt 1 of 3): Filter@FrameProcessor: config item NoiseRemovalFilter#1 doesn't exist
```

I am on the latest 1.1.2 firmware.

christian-rauch, 2025-05-23:
The last working commit is 10aebfd89632fbe4cb85d9dae95737c9b96c510c. The issue appears after `Update sdk version to v2.4.1` (b2ef7d2d07419c2a3d6539e2cb2ccadb353e3ab9). So this seems to be related to the binary SDK files in this repo.

@jjiszjj Can you take care of this and revert the breaking changes?

mpowelson, 2025-07-31:
I experienced the same problem with the 335Le. Removing `--symlink-install` as suggested in #80 fixed it for me.

Danilrivero, 2025-09-29:
Hi, experienced the same issue when migrating to `2.4.5`.

Is there any additional info. on how to tackle this?