orbbec/OrbbecSDK_ROS2 #121: Get small colored point cloud while still get full sized color image?

maierp, 2025-06-16:
Is there a way to get the small colored point cloud from the 640x576 depth image while still get the large RGB image of 3840x2160?

When I try to get the colored point cloud, the depth image send to be scaled up to fit the color image and then a huge point cloud is generated. 

When I try to set the align_mode to depth, I get the small point cloud, but the RGB image is also scaled down to fit the depth image and it's also transformed.

I cannot use the down scaled and depth aligned color image, because I want to run a marker tracker on the color image. 

Thank you for any advice ☺️

jjiszjj, 2025-06-19:
@maierpOh, I misunderstood what you meant. I thought you wanted to output a 640x576 depth point cloud, but you actually wanted to output a color point cloud.

This requires secondary development of OrbbecSDK_ROS2 based on setting the align_mode to depth, and publishing the color topic before aligning to depth for 3840x2160 Color.

jjiszjj, 2025-06-19:
@maierp You can publish the 3840x2160 color separately before the [align filter](https://github.com/orbbec/OrbbecSDK_ROS2/blob/e52169a56f97efa1d05e2b7db5530d324f764061/orbbec_camera/src/ob_camera_node.cpp#L2554)

maierp, 2025-06-19:
Ah cool I'll try to implement this. Thanks a lot for the help ☺️👍

closed 2025-10-22 by obyalian