orbbec/OrbbecSDK_ROS2 #136: Femto Bolt not launching

jayant-mil, 2025-08-19:
When I try to launch femto bolt with `ros2 launch  orbbec_camera femto_bolt.launch.py`, it fails to launch giving the following error:

```sh
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [11297]
[component_container-1] [INFO] [1755594275.464164840] [camera.camera_container]: Load Library: /home/milaipc/ros2ws_isortx_robot/install/orbbec_camera/lib/liborbbec_camera.so
[component_container-1] [INFO] [1755594275.498060649] [camera.camera_container]: Found class: rclcpp_components::NodeFactoryTemplate<orbbec_camera::OBCameraNodeDriver>
[component_container-1] [INFO] [1755594275.498110094] [camera.camera_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<orbbec_camera::OBCameraNodeDriver>
[component_container-1] [08/19 11:04:35.500223][info][11297][Context.cpp:68] Context created with config: /home/milaipc/ros2ws_isortx_robot/install/orbbec_camera/share/orbbec_camera/config/OrbbecSDKConfig_v1.0.xml
[component_container-1] [08/19 11:04:35.500229][info][11297][Context.cpp:73] Work directory=/home/milaipc/ros2ws_isortx_robot, SDK version=v1.10.22-20250410-46139de
[component_container-1] [08/19 11:04:35.500234][info][11297][LinuxPal.cpp:32] createObPal: create LinuxPal!
[component_container-1] [08/19 11:04:35.763015][info][11297][LinuxPal.cpp:166] Create PollingDeviceWatcher!
[component_container-1] [08/19 11:04:35.763049][info][11297][DeviceManager.cpp:15] Current found device(s): (3)
[component_container-1] [08/19 11:04:35.763050][info][11297][DeviceManager.cpp:24] 	- Name: Orbbec Gemini 2, PID: 0x0670, SN/ID: AY3794301A0, Connection: USB3.0
[component_container-1] [08/19 11:04:35.763051][info][11297][DeviceManager.cpp:24] 	- Name: Orbbec Gemini 2, PID: 0x0670, SN/ID: AY3794301C4, Connection: USB3.0
[component_container-1] [08/19 11:04:35.763052][info][11297][DeviceManager.cpp:24] 	- Name: Femto Bolt, PID: 0x066b, SN/ID: CL8855301KF, Connection: USB3.1
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/camera/camera' in container '/camera/camera_container'
[component_container-1] [INFO] [1755594275.864089574] [camera.camera]: Connecting to the default device
[component_container-1] [INFO] [1755594276.023205127] [camera.camera]: Select device cost 159 ms
[component_container-1] [INFO] [1755594276.023261921] [camera.camera]: Try to connect device via USB3.0
[component_container-1] [INFO] [1755594276.023317111] [camera.camera]: OBCameraNode: use_intra_process: OFF
[component_container-1] [INFO] [1755594276.025556747] [camera.camera]: current time domain: device
[component_container-1] [INFO] [1755594276.034694310] [camera.camera]: Setting heartbeat to OFF
[component_container-1] [INFO] [1755594276.035119702] [camera.camera]: Depth process is HW D2D
[component_container-1] [INFO] [1755594276.035131099] [camera.camera]: Setting LDP to ON
[component_container-1] [INFO] [1755594276.035738919] [camera.camera]: Current sync mode: OB_MULTI_DEVICE_SYNC_MODE_STANDALONE
[component_container-1] [INFO] [1755594276.038734144] [camera.camera]: Set sync mode: OB_MULTI_DEVICE_SYNC_MODE_STANDALONE
[component_container-1] [INFO] [1755594276.038744007] [camera.camera]: Setting depth soft filter to ON
[component_container-1] [INFO] [1755594276.038756426] [camera.camera]: Setting color auto white balance to ON
[component_container-1] [INFO] [1755594276.038825437] [camera.camera]: Setting color auto exposure to ON
[component_container-1] [INFO] [1755594276.040608856] [camera.camera]: Setting IR auto exposure to ON
[component_container-1] [INFO] [1755594276.045267759] [camera.camera]: default_soft_filter_max_diff: 64
[component_container-1] [INFO] [1755594276.045284547] [camera.camera]: default_soft_filter_speckle_size: 200
[component_container-1] [INFO] [1755594276.046125566] [camera.camera]: Setting EdgeNoiseRemovalFilter......
[component_container-1] [INFO] [1755594276.046138955] [camera.camera]: Skip setting filter: EdgeNoiseRemovalFilter
[component_container-1] [INFO] [1755594276.046146844] [camera.camera]: Setting SpatialAdvancedFilter......
[component_container-1] [INFO] [1755594276.046152804] [camera.camera]: set SpatialAdvancedFilter to false
[component_container-1] [INFO] [1755594276.046173693] [camera.camera]: Skip setting filter: SpatialAdvancedFilter
[component_container-1] [INFO] [1755594276.046180906] [camera.camera]: Setting TemporalFilter......
[component_container-1] [INFO] [1755594276.046186437] [camera.camera]: set TemporalFilter to false
[component_container-1] [INFO] [1755594276.046192247] [camera.camera]: Skip setting filter: TemporalFilter
[component_container-1] [INFO] [1755594276.046198642] [camera.camera]: Setting HoleFillingFilter......
[component_container-1] [INFO] [1755594276.046203981] [camera.camera]: set HoleFillingFilter to false
[component_container-1] [INFO] [1755594276.046209445] [camera.camera]: Skip setting filter: HoleFillingFilter
[component_container-1] [INFO] [1755594276.047299101] [camera.camera]:  stream color is enabled - width: 1280, height: 720, fps: 30, Format: OB_FORMAT_MJPG
[component_container-1] [ERROR] [1755594276.047415076] [camera.camera]: Failed to get depth  profile: Invalid input, No matched video stream profile found!
[component_container-1] [ERROR] [1755594276.047429256] [camera.camera]: Stream: OB_STREAM_DEPTH, Stream Index: 0, Width: 640, Height: 576, FPS: 30, Format: OB_FORMAT_Y16
[component_container-1] [ERROR] [1755594276.047432158] [camera.camera]: Error: The device might be connected via USB 2.0. Please verify your configuration and try again. The current process will now exit.
[component_container-1] [INFO] [1755594276.047434809] [camera.camera]: Available profiles:
[component_container-1] [INFO] [1755594276.047440794] [camera.camera]: depth profile: 1280x800 30fps REL
[component_container-1] [INFO] [1755594276.047443316] [camera.camera]: depth profile: 320x200 30fps Y14
[component_container-1] [INFO] [1755594276.047445035] [camera.camera]: depth profile: 320x200 15fps Y14
[component_container-1] [INFO] [1755594276.047446643] [camera.camera]: depth profile: 320x200 10fps Y14
[component_container-1] [INFO] [1755594276.047448476] [camera.camera]: depth profile: 320x200 5fps Y14
[component_container-1] [INFO] [1755594276.047450052] [camera.camera]: depth profile: 640x400 30fps Y14
[component_container-1] [INFO] [1755594276.047451679] [camera.camera]: depth profile: 640x400 15fps Y14
[component_container-1] [INFO] [1755594276.047453190] [camera.camera]: depth profile: 640x400 10fps Y14
[component_container-1] [INFO] [1755594276.047455029] [camera.camera]: depth profile: 640x400 5fps Y14
[component_container-1] [INFO] [1755594276.047456592] [camera.camera]: depth profile: 1280x800 30fps Y14
[component_container-1] [INFO] [1755594276.047458301] [camera.camera]: depth profile: 1280x800 15fps Y14
[component_container-1] [INFO] [1755594276.047460131] [camera.camera]: depth profile: 1280x800 10fps Y14
[component_container-1] [INFO] [1755594276.047462003] [camera.camera]: depth profile: 1280x800 5fps Y14
[component_container-1] [INFO] [1755594276.047463616] [camera.camera]: depth profile: 320x200 30fps REL
[component_container-1] [INFO] [1755594276.047465224] [camera.camera]: depth profile: 320x200 15fps REL
[component_container-1] [INFO] [1755594276.047466851] [camera.camera]: depth profile: 320x200 10fps REL
[component_container-1] [INFO] [1755594276.047468497] [camera.camera]: depth profile: 320x200 5fps REL
[component_container-1] [INFO] [1755594276.047470083] [camera.camera]: depth profile: 640x400 30fps REL
[component_container-1] [INFO] [1755594276.047471692] [camera.camera]: depth profile: 640x400 15fps REL
[component_container-1] [INFO] [1755594276.047473268] [camera.camera]: depth profile: 640x400 10fps REL
[component_container-1] [INFO] [1755594276.047475084] [camera.camera]: depth profile: 640x400 5fps REL
[... 48 lines cut ...]
[ 1873.035617] usb 1-7.1: Failed to query (GET_CUR) UVC control 5 on unit 3: -71 (exp. 1).
[ 1873.035742] usb 1-7.1: Failed to query (GET_INFO) UVC control 11 on unit 3: -71 (exp. 1).
[ 1873.035866] usb 1-7.1: Failed to query (GET_INFO) UVC control 2 on unit 1: -71 (exp. 1).
[ 1873.035992] usb 1-7.1: Failed to query (GET_INFO) UVC control 4 on unit 1: -71 (exp. 1).
[ 1873.036116] usb 1-7.1: Failed to query (GET_INFO) UVC control 15 on unit 1: -71 (exp. 1).
[ 1873.037083] usb 1-7.1: UVC non compliance - GET_DEF(PROBE) not supported. Enabling workaround.
[... cut ...]
[ 1876.185228] usb 2-7.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1876.185229] usb 2-7.1: Product: Orbbec Femto Bolt 3D Camera
[ 1876.185230] usb 2-7.1: Manufacturer: Orbbec
[ 1876.185230] usb 2-7.1: SerialNumber: CL8855301KF
[ 1876.187783] usb 2-7.1: Found UVC 1.00 device Orbbec Femto Bolt 3D Camera (2bc5:066b)
[ 1876.195084] usb 2-7.1: Found UVC 1.00 device Orbbec Femto Bolt 3D Camera (2bc5:066b)
[ 1876.197723] hid-generic 0003:2BC5:066B.001A: hiddev3,hidraw7: USB HID v1.01 Device [Orbbec Orbbec Femto Bolt 3D Camera] on usb-0000:00:14.0-7.1/input4
```

obyalian, 2025-08-20:
It looks like the issue is caused by having multiple cameras connected at the same time, while the launch file did not explicitly specify which device to use. In that case, the driver will connect to the default device.

You can check the serial number of your Femto Bolt by running:
```bash
ros2 run orbbec_camera list_devices_node
```

Then launch with the serial number explicitly set, for example:
```bash 
ros2 launch orbbec_camera femto_bolt.launch.py serial_number:=CL8H741005J
```

jayant-mil, 2025-08-20:
Perfect! That works, thanks!