orbbec/OrbbecSDK_ROS2 #139: Femto Bolt, URDF model wrong (?)

fabioscap, 2025-08-22:
Hello,

I am suspecting that there is something wrong the Femto Bolt description. I am running an Ouster LiDAR and a Femto Bolt on the same robot. Here is my sensor setup:

[image]

Here is the URDF model:

[image]

By comparing the clouds from the LiDAR and the Bolt, I am expecting a small misalignment due to my limited precision in mounting the sensor. As a matter of fact there is: 

[image]

So if I register the two scans I can get the actual offset between the LiDAR and Bolt:

[image]

But then the transform on the URDF looks way off and corrects way more than what is reasonably expected from human precision. There is no way the camera sits so far back:

[image]

I cannot really understand where my mistake could be. I made sure to disambiguate other sources of problems. 
- done the LiDAR-Bolt extrinsic calibration at many different depth ranges, reliably obtaining the same offset.
- also, there is a robotic arm in front of the camera. You can see it in the white patch in the last screenshot. That patch also gets aligned (visually) to the actual arm when correcting the extrinsic. so I am tempted to exclude the LiDAR from the problem.
- the depth and color frames are aligned with each other as far as I can see plotting the colored point cloud.
- I am using branch v2-main
- I have tried undoing and doubling the tilt described in #44, #52 and related issues but that did not help. Not one single rotation can correct the cm-scale offset I have.

As a temporary solution, I have added a transform that corrects the visual model depth frame to the actual depth frame. 
Here is the approximate transform:
```
  <link name="camera_link"/>
  <joint name="camera_visual_to_camera" type="fixed">
    <parent link="camera_visual_link"/>
    <child link="camera_link"/>
    <!-- <origin xyz="0.0 0.0 0.0" rpy="0.0 0.0 0.0"/> -->
    <origin xyz="-0.029750074843318527 -0.005784185950938374 -0.05113393129377081" rpy="-0.018286819223818494 0.017199897606273495 0.010042825389099981"/>
  </joint>
```

However, I would like to understand the root cause of this problem so it would be helpful if you could double check the URDF model or suggest me where the error might be. Thanks,
fabio

christian-rauch, 2025-08-27:
Are you using the correct frames? I have attached the Femto Bolt, wrist-mounted, at the end of a robot arm. Using the forward kinematics of the arm, the CAD information from the mounting jig, and the TF from the camera model, I get a pretty good alignment of the point clouds when moving the arm around.

I would visualise the frames of your TF tree and see if they match with the real world. Then I would check if the colour and the registered depth image are given in the correct optical frame.

I noticed that some frames in the model are wrong. The IMU frame, for example, is outside the model: https://github.com/orbbec/OrbbecSDK_ROS2/issues/103. But the frames for the cameras seem to be correct.

fabioscap, 2025-09-01:
Hello, thanks for the reply!
I would like to think that I've used the correct frames, but I am going to triple check. 
Also, I don't know if it's relevant, but I cannot reproduce #103. Here is what my tfs look like (I am using `use_nominal_extrinsics="false"` in the URDF and `enable_publish_extrinsic: true` in the launch file):

```
camera_link ->  camera_gyro_frame
- Translation: [-0.053, -0.038, 0.010]
- Rotation: in Quaternion [0.499, 0.498, -0.502, 0.501]
```

```
camera_link ->  camera_accel_frame
- Translation: [-0.053, -0.038, 0.010]
- Rotation: in Quaternion [0.499, 0.498, -0.502, 0.501]
```
[image]

christian-rauch, 2025-09-01:
It looks like the issue I reported is only present with the v1 on the `main` branch and not v2 on the `v2-main` branch.

obyalian, 2025-09-08:
Got it, thanks for letting us know! From our preliminary analysis, the URDF we provided does have some issues, although they may not be the direct cause of your mismatch. We’ll work on fixing it as soon as possible. Thanks again for your feedback!

Daongvan02, 2026-08-15:
@ob-yalian I wanted you to update the Femto Mega URDF model